platformio/platformio-core #4604: Issue with multiple targets

fabianoriccardi, 2023-04-22:
### Bug

### Configuration

**Operating system**: Windows 10

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 6.1.6

### Description of problem

In platformio.ini I have created an env with the following target in order to flash an ESP8266:

    [env:my_env]
    targets = clean, upload, uploadfs, monitor

(other options are omitted for brevity)

I would expect they run sequentially, instead they run in parallel. Running in parallel, they lead to an obvious error since the last 3 targets try to acquire the serial port all together (the fastest target is `monitor`, so in the end `upload` and `uploadfs` fail).

### Actual Results

The targets run in parallel.

### Expected Results

The targets run sequentially.

### Additional info

Issue already posted and confirmed on PIO forum [here](https://community.platformio.org/t/issue-with-multiple-targets/33372).

ivankravets, 2023-04-25:
Thanks for the report. Please run `pio upgrade --dev` command and re-test. It should work now.

closed 2023-04-25 by ivankravets

fabianoriccardi, 2023-04-25:
Now it **partially** works!

If I run 

    pio run -t clean -t upload -t uploadfs -t monitor

it uploads 2 times the fs, but no firmware! Clean and monitor are executed properly.

If I run 

    pio run -t clean -t upload -t monitor

it uploads the firmware as expected.

pio --version gives `6.1.7b2`

ivankravets, 2023-04-26:
@fabianoriccardi now, this is the issue of epsressif32 dev-platform. Could you report it to https://github.com/platformio/platform-espressif32/issues ?

Thanks in advance!

ivankravets, 2023-04-26:
@vshymanskyy , could you re-test the latest development PIO Core (`pio upgrade --dev`) for your workflow with multiple targets?

fabianoriccardi, 2023-04-26:
I'm working on ESP8266, so I have opened the issue there.

commit 6f0b1fbb (2023-04-25): Resolved an issue where multiple targets were not executed sequentially // Resolve #4604