platformio/platformio-core #4606: Resolving nodemcu-32s dependencies... fails

ZZKK000, 2023-04-25:
------------------------------------------------------------------

### Configuration

**Operating system**: win10

**PlatformIO Version** (`platformio --version`): 6.1.6-->6.1.7b2

### Description of problem

when load project for ESP32 at arduino framework, error happens.

********************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
********************************************************************************

Resolving nodemcu-32s dependencies...
Removing unused dependencies...
Error: Traceback (most recent call last):
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\platformio\__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\platformio\cli.py", line 70, in invoke
    return super().invoke(ctx)
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\platformio\project\commands\init.py", line 92, in project_init_cmd
    install_project_dependencies(
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 106, in install_project_dependencies
    already_up_to_date = not install_project_env_dependencies(env, options)
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 132, in install_project_env_dependencies
    _install_project_env_libraries(project_env, options),
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 204, in _install_project_env_libraries
    _uninstall_project_unused_libdeps(project_env, options)
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 280, in _uninstall_project_unused_libdeps
    lm.uninstall(spec)
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\platformio\package\manager\_uninstall.py", line 29, in uninstall
    return self._uninstall(spec, skip_dependencies)
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\platformio\package\manager\_uninstall.py", line 34, in _uninstall
    pkg = self.get_package(spec)
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\platformio\package\manager\base.py", line 263, in get_package
    spec = self.ensure_spec(spec)
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\platformio\package\manager\base.py", line 133, in ensure_spec
    return spec if isinstance(spec, PackageSpec) else PackageSpec(spec)
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 183, in __init__
    self._parse(self.raw)
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 290, in _parse
    raw = parser(raw)
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 315, in _parse_requirements
    self.requirements = tokens[1].strip()
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 230, in requirements
    else semantic_version.SimpleSpec(str(value))
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\semantic_version\base.py", line 647, in __init__
    self.clause = self._parse_to_clause(expression)
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\semantic_version\base.py", line 1043, in _parse_to_clause
    return cls.Parser.parse(expression)
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\semantic_version\base.py", line 1063, in parse
    raise ValueError("Invalid simple block %r" % block)
ValueError: Invalid simple block '^2.0.'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

#### Steps to Reproduce

1. pip install -U platformio
2. pio system prune
3. pio upgrade --dev

### Actual Results

not solved 

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**

[env:nodemcu-32s]
platform = espressif32
board = nodemcu-32s
framework = arduino
monitor_speed = 115200
lib_deps = knolleary/PubSubClient@^2.8

**Source file to reproduce issue:**

#include <Arduino.h>

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
### Additional info

ZZKK000, 2023-04-25:
after deleting all dependencies from the platformio.ini, error message disappears:

Resolving nodemcu-32s dependencies...
Already up-to-date.
Updating metadata for the vscode IDE...
Project has been successfully updated!

ivankravets, 2023-04-25:
Please reopen if you still need help

ZZKK000, 2023-04-25:
how to reopen this topic?

ZZKK000, 2023-04-25:
> Please reopen if you still need help

how to reopen this topic?

ivankravets, 2023-04-25:
Could you provide a simple project to reproduce this issue?

ZZKK000, 2023-04-25:
> Could you provide a simple project to reproduce this issue?

Yes, I just created a new blank project,  after added a library, the error message appeared immediately again.

main.cpp is blank as below:
**************************************************************************

#include <Arduino.h>

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}

****************************************************************************************

add a library in plateformio.ini as below:

*************************************************************************************
[env:nodemcu-32s]
board_build.mcu = esp32
board_build.f_cpu = 240000000L
platform = espressif32
board = node32s
framework = arduino
monitor_speed = 115200
lib_deps = knolleary/PubSubClient@^2.8

****************************************************************************************

error message again in output console window:

Resolving nodemcu-32s dependencies...
Removing unused dependencies...
Error: Traceback (most recent call last):
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\platformio\__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\platformio\cli.py", line 70, in invoke
    return super().invoke(ctx)
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\platformio\project\commands\init.py", line 92, in project_init_cmd
    install_project_dependencies(
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 106, in install_project_dependencies
    already_up_to_date = not install_project_env_dependencies(env, options)
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 132, in install_project_env_dependencies
    _install_project_env_libraries(project_env, options),
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 204, in _install_project_env_libraries
    _uninstall_project_unused_libdeps(project_env, options)
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 280, in _uninstall_project_unused_libdeps
    lm.uninstall(spec)
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\platformio\package\manager\_uninstall.py", line 29, in uninstall
    return self._uninstall(spec, skip_dependencies)
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\platformio\package\manager\_uninstall.py", line 34, in _uninstall
    pkg = self.get_package(spec)
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\platformio\package\manager\base.py", line 263, in get_package
    spec = self.ensure_spec(spec)
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\platformio\package\manager\base.py", line 133, in ensure_spec
    return spec if isinstance(spec, PackageSpec) else PackageSpec(spec)
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 183, in __init__
    self._parse(self.raw)
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 290, in _parse
    raw = parser(raw)
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 315, in _parse_requirements
    self.requirements = tokens[1].strip()
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 230, in requirements
    else semantic_version.SimpleSpec(str(value))
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\semantic_version\base.py", line 647, in __init__
    self.clause = self._parse_to_clause(expression)
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\semantic_version\base.py", line 1043, in _parse_to_clause
    return cls.Parser.parse(expression)
  File "C:\Users\HASEE\.platformio\penv\lib\site-packages\semantic_version\base.py", line 1063, in parse
    raise ValueError("Invalid simple block %r" % block)
ValueError: Invalid simple block '^2.0.'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

maxgerhardt, 2023-04-25:
Does it also appear without the `lib_deps ` line?

ZZKK000, 2023-04-25:
> Does it also appear without the `lib_deps ` line?

no

maxgerhardt, 2023-04-25:
Weird, `pio pkg install -g -l "knolleary/PubSubClient@^2.8"` goes through without any errors for me. Can you open a [CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) and execute the same command?

ZZKK000, 2023-04-25:
> Weird, `pio pkg install -g -l "knolleary/PubSubClient@^2.8"` goes through without any errors for me. Can you open a [CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) and execute the same command?

l will try in next day, thanks!

ZZKK000, 2023-04-26:
> Weird, `pio pkg install -g -l "knolleary/PubSubClient@^2.8"` goes through without any errors for me. Can you open a [CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) and execute the same command?

pio pkg install -g -l "knolleary/PubSubClient@^2.8"
this command line is effective. loading and compling project all become OK. However, once using "lib_deps = " again in platformio.ini , error occurs again.

ZZKK000, 2023-04-26:
> Please remove `C:\Users\HASEE.platformio\penv` folder and restart VSCode.

Hello, ivankravets:

This method didn't work.

ZZKK000, 2023-04-26:
> Weird, `pio pkg install -g -l "knolleary/PubSubClient@^2.8"` goes through without any errors for me. Can you open a [CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) and execute the same command?

With the command line "pio pkg insall -g -l" ,  all libraries have been imported into this folder: 
C:\Users\HASEE\.platformio\lib, so there is no need of "iib_deps=", 
but it is not convenient to search and import the libs from PIO Home libraries window.

maxgerhardt, 2023-04-26:
The `-g` in the command means *global*. That wa just a test. Delete evrything in `C:\Users\HASEE\.platformio\lib` again, delete the `.pio` folder of your project rebuild intellisense (Ctrl+Shift+P -> Rebuild Intellisense), with the `lib_deps` line still being in the project.

ZZKK000, 2023-04-26:
> The `-g` in the command means _global_. That wa just a test. Delete evrything in `C:\Users\HASEE\.platformio\lib` again, delete the `.pio` folder of your project rebuild intellisense (Ctrl+Shift+P -> Rebuild Intellisense), with the `lib_deps` line still being in the project.

Hello, maxgerhardt:

This method works, and also seems no problem for creating a new project with "lib_deps". Thanks a lot!

maxgerhardt, 2023-04-26:
Great, then I think there was just some weird residue in the `.pio` folder with a malformed dependency string.

ivankravets, 2023-04-26:
@maxgerhardt , thanks for the help! 🙏

closed 2023-04-26 by ivankravets