platformio/platformio-core #4607: pio remote agent throw's exception on raspberry pi's (pio core version 6.1.7b2)

mrcamilletti, 2023-04-25:
I'm having a similar issue as this one (https://github.com/platformio/platformio-core/issues/3652#issue-693508852) using the new version (core version 6.1.7b2). Remote agent was working fine before, and after some days (an update I suppose) this happened.

### Configuration
```
$ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

$ pio system info
--------------------------  -----------------------------------------
PlatformIO Core             6.1.7b2
Python                      3.7.3-final.0
System Type                 linux_armv7l
Platform                    Linux-5.10.103-v7l+-armv7l-with-glibc2.28
File System Encoding        utf-8
Locale Encoding             UTF-8
PlatformIO Core Directory   /home/pi/.platformio
PlatformIO Core Executable  /home/pi/.platformio/penv/bin/platformio
Python Executable           /home/pi/.platformio/penv/bin/python
Global Libraries            0
Development Platforms       0
Tools & Toolchains          1
--------------------------  -----------------------------------------
```

### Description
After running `pio remote agent start` the process tries to install libraries and fails running `Collecting attrs>=19.2.0 (from twisted==22.1.0)`. 

### Steps to reproduce
- Install required libraries: $ sudo apt install python3-dev libffi-dev libssl-dev
- Clean install of platformio core using installer-script [Super-Quick (macOS / Linux)](https://docs.platformio.org/en/latest/core/installation/methods/installer-script.html#id1)
- pio upgrade --dev
- pio account login
- pio remote agent start

### Result
```
Please wait while upgrading PlatformIO...
PlatformIO has been successfully upgraded to 6.1.7b2!

**********************************************************************************************************************************************************************************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for embedded development > https://platformio.org/platformio-ide
**********************************************************************************************************************************************************************************************************************************************

Tool Manager: Installing platformio/contrib-pysite @ ~2.37.0
Unpacking  [####################################]  100%          
Tool Manager: contrib-pysite@2.37.191020 has been installed!
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting twisted==22.1.0
  Using cached Twisted-22.1.0-py3-none-any.whl
Collecting cryptography<35.0.0,>=3.3
  Using cached cryptography-3.4.8-cp37-cp37m-linux_armv7l.whl
Collecting pyopenssl<=21.0.0,>=16.0.0
  Using cached pyOpenSSL-21.0.0-py2.py3-none-any.whl
Collecting service_identity<=21.1.0,>=18.1.0
  Using cached service_identity-21.1.0-py2.py3-none-any.whl
Collecting zope.interface>=4.4.2 (from twisted==22.1.0)
  Using cached zope.interface-6.0.tar.gz (246 kB)
  Preparing metadata (setup.py) ... done
[... 212 lines cut ...]
      
        note: This error originates from a subprocess, and is likely not a problem with pip.
      error: subprocess-exited-with-error
      
      × pip subprocess to install backend dependencies did not run successfully.
      │ exit code: 1
      ╰─> See above for output.
      
      note: This error originates from a subprocess, and is likely not a problem with pip.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
Error: 

Please ensure that the next packages are installed:

sudo apt install python3-dev libffi-dev libssl-dev
```

maxgerhardt, 2023-04-25:
May be related: https://github.com/pypa/pip/issues/5739 and https://github.com/pypa/pip/issues/11363

ivankravets, 2023-04-27:
Now dependencies will be built on the target side. Please re-test with `pio upgrade --dev`. Does it work now?

closed 2023-04-27 by ivankravets

mrcamilletti, 2023-04-27:
> Now dependencies will be built on the target side. Please re-test with `pio upgrade --dev`. Does it work now?

Yes! it works now, thanks!

commit 2c94fb2a (2023-04-27): Resolved installation issues with PIO Remote on Raspberry Pi and other small form-factor PCs // Resolve #4425 , Resolve #4493 , Resolve  #4607