platformio/platformio-core #4609: Teknic Clearcore support

jnowak1054, 2023-04-27:
- [ ] Development Platform or Board.

First thing's first. Thanks for this wonderful project.

Would like you to add support for Teknic ClearCore (based off of a SAME53N19A) link: https://teknic.com/products/io-motion-controller/
They have multiple repositories here https://github.com/Teknic-Inc

It seems to me an excellent low-cost option for industrial applications based on the Arduino ecosystem, industrial grade I/O with robust motion control capabilities.

I have one that I can make available for any necessary test.

Thank you very much in advance.

NWClimber, 2023-05-06:
Big upvote for ClearCore. That's why I'm here.

ivankravets, 2023-05-07:
Please press 👍  button in the original comment above.

patrickwasp, 2023-07-06:
https://github.com/platformio/platform-atmelsam/issues/175

patrickwasp, 2023-08-29:
> With the help of @maxgerhardt I added support for the ClearCore.
> 
> While the pull request is being considered you can use your ClearCore in PlatformIO by using the following config:
> 
> ```
> [env:clearcore]
> platform = https://github.com/patrickwasp/platform-atmelsam
> board = clearcore
> framework = arduino
> ```

jnowak1054, 2023-08-29:
Awesome! Thank you!