platformio/platformio-core #4614: platformio fails to run on macOS Ventura due to urllib3

famousmoose, 2023-05-06:
- [ X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Mac OS Ventura 13.3.1 (a) ARM64

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 6.1.6

### Description of problem
A default install of platformio on a MacOS host using the default python and python ssl module will fail to run, giving the following error
```
$ platformio --version
Traceback (most recent call last):
  File "/Users/stu/.platformio/penv/bin/platformio", line 5, in <module>
    from platformio.__main__ import main
  File "/Users/stu/.platformio/penv/lib/python3.9/site-packages/platformio/__main__.py", line 21, in <module>
    from platformio import __version__, exception, maintenance
  File "/Users/stu/.platformio/penv/lib/python3.9/site-packages/platformio/maintenance.py", line 22, in <module>
    from platformio import __version__, app, exception, fs, telemetry
  File "/Users/stu/.platformio/penv/lib/python3.9/site-packages/platformio/telemetry.py", line 28, in <module>
    import requests
  File "/Users/stu/.platformio/penv/lib/python3.9/site-packages/requests/__init__.py", line 43, in <module>
    import urllib3
  File "/Users/stu/.platformio/penv/lib/python3.9/site-packages/urllib3/__init__.py", line 38, in <module>
    raise ImportError(
ImportError: urllib3 v2.0 only supports OpenSSL 1.1.1+, currently the 'ssl' module is compiled with LibreSSL 2.8.3. See: https://github.com/urllib3/urllib3/issues/2168
```

This is due to the recently launched `urllib3` version 2 series throws errors when used with libressl and older versions of openssl. urllib3 is a dependency for `requests`

#### Steps to Reproduce

1. Install platfromio using platformio installer (e.g. `python3 -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/master/scripts/get-platformio.py)"`
2. run `platformio --version`

### Actual Results
The error listed above

### Expected Results
PlatformIO Core, version 6.1.6

### Additional info
A workaround is as follows:
```
. ~/.platformio/penv/bin/activate
pip install urllib3==1.26.15
```

This allows the platformio command to work correctly.

ivankravets, 2023-05-06:
Thanks for the report!

closed 2023-05-06 by ivankravets

ivankravets, 2023-05-08:
Thanks for pinging. 6.1.7 is ready! https://github.com/platformio/platformio-core/releases/tag/v6.1.7

🚀

nstansbury, 2023-09-20:
For others coming here, you will now get a warning even with `6.1.11` due to a change in `urllib3` allowing support for LibreSSL with a warning.  This occurs despite using the PIO installed Python and having `openssl@1.1.1d` installed and in my `PATH$`

```
/Volumes/DATA/PlatformIO/penv/lib/python3.8/site-packages/urllib3/__init__.py:34: NotOpenSSLWarning: urllib3 v2.0 only supports OpenSSL 1.1.1+, currently the 'ssl' module is compiled with 'LibreSSL 2.8.3'. See: https://github.com/urllib3/urllib3/issues/3020 warnings.warn(
```

```
PlatformIO Core             6.1.11
Python                      3.8.2-final.0
System Type                 darwin_x86_64
Platform                    macOS-10.15.7
File System Encoding        utf-8
Locale Encoding             UTF-8
PlatformIO Core Directory   /Volumes/DATA/PlatformIO
PlatformIO Core Executable  /Volumes/DATA/PlatformIO/penv/bin/platformio
Python Executable           /Volumes/DATA/PlatformIO/penv/bin/python
Global Libraries            0
Development Platforms       2
Tools & Toolchains          9
```

"Fixed" by following [maxgerhardt](https://github.com/maxgerhardt) suggestion to downgrade `urllib3` to `1.26.15`

lemonkey, 2023-11-23:
Also noticed this started happening for me after I upgraded to macOS Sonoma 14.0.

Also "Fixed" by following [maxgerhardt](https://github.com/maxgerhardt) suggestion to downgrade urllib3 to 1.26.15

Note that I had to first change into ~/.platformio/penv and then run `source bin/activate`.

commit bba5bc9d (2023-05-06): Lock "urllib3<2" becuase urllib3 v2.0 only supports OpenSSL 1.1.1+ // Resolve #4614