platformio/platformio-core #4620: Update broke compling

fabltd, 2023-05-11:
When complying PI 2040 target with Ardunio core the following occours:

```
In file included from .pio/libdeps/prod/WebSerial/src/WebSerial.cpp:1:
.pio/libdeps/prod/WebSerial/src/WebSerial.h:5:10: fatal error: stdlib_noniso.h: No such file or directory

***********************************************************************
* Looking for stdlib_noniso.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:stdlib_noniso.h"
* Web  > https://registry.platformio.org/search?q=header:stdlib_noniso.h
*
***********************************************************************

    5 | #include "stdlib_noniso.h"
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/prod/lib8a1/WebSerial/WebSerial.cpp.o] Error 1
```

This has been working fine and appears not to be specfic to the 2040 core. 

Help

fabltd, 2023-05-11:
Ok not a PIO Core issue:

This lib is broken: 
`ayushsharma82/WebSerial
`
I did not need to use it so removed and back on track.

closed 2023-05-11 by fabltd

ivankravets, 2023-05-11:
Happy coding with PlatformIO! 😊