platformio/platformio-core #4623: Failing tests on a freshly downloaded project

MacDada, 2023-05-12:
- [x] **PlatformIO Core**.

**Operating system**: `MacOS 12.6.5`

**PlatformIO Version** (`platformio --version`): `PlatformIO Core, version 6.1.8a1`

First thing, [I do have Python 3.9, not Python 3.7](https://github.com/platformio/platformio-core/issues/4622) – I don't know if that makes any difference here.

Per https://github.com/platformio/platformio-core/blob/develop/CONTRIBUTING.md
I'm trying to run tests and they fail:

```
(py39) ➜  platformio-core git:(develop) make test 

py.test --verbose --exitfirst -n 6 --dist=loadscope tests --ignore tests/test_examples.py
========================================= test session starts =========================================
platform darwin -- Python 3.9.16, pytest-7.3.1, pluggy-1.0.0 -- /Users/dada/Documents/projects/forks/platformio-core/.tox/py39/bin/python
cachedir: .pytest_cache
rootdir: /Users/dada/Documents/projects/forks/platformio-core
configfile: tox.ini
plugins: xdist-3.2.1, anyio-3.6.2
[gw0] darwin Python 3.9.16 cwd: /Users/dada/Documents/projects/forks/platformio-core
[gw1] darwin Python 3.9.16 cwd: /Users/dada/Documents/projects/forks/platformio-core
[gw2] darwin Python 3.9.16 cwd: /Users/dada/Documents/projects/forks/platformio-core
[gw3] darwin Python 3.9.16 cwd: /Users/dada/Documents/projects/forks/platformio-core
[gw4] darwin Python 3.9.16 cwd: /Users/dada/Documents/projects/forks/platformio-core
[gw5] darwin Python 3.9.16 cwd: /Users/dada/Documents/projects/forks/platformio-core
[gw0] Python 3.9.16 (main, Dec  7 2022, 10:15:43)  -- [Clang 14.0.0 (clang-1400.0.29.202)]
[gw1] Python 3.9.16 (main, Dec  7 2022, 10:15:43)  -- [Clang 14.0.0 (clang-1400.0.29.202)]
[gw2] Python 3.9.16 (main, Dec  7 2022, 10:15:43)  -- [Clang 14.0.0 (clang-1400.0.29.202)]
[gw3] Python 3.9.16 (main, Dec  7 2022, 10:15:43)  -- [Clang 14.0.0 (clang-1400.0.29.202)]
[gw4] Python 3.9.16 (main, Dec  7 2022, 10:15:43)  -- [Clang 14.0.0 (clang-1400.0.29.202)]
[gw5] Python 3.9.16 (main, Dec  7 2022, 10:15:43)  -- [Clang 14.0.0 (clang-1400.0.29.202)]
gw0 [225] / gw1 [225] / gw2 [225] / gw3 [225] / gw4 [225] / gw5 [225]
scheduling tests via LoadScopeScheduling

tests/commands/test_account_org_team.py::test_account_register 
[gw0] [  0%] SKIPPED tests/commands/test_account_org_team.py::test_account_register 
tests/commands/test_account_org_team.py::test_account_login 
[gw0] [  0%] SKIPPED tests/commands/test_account_org_team.py::test_account_login 
tests/commands/test_account_org_team.py::test_account_summary 
tests/commands/test_boards.py::test_board_json_output 
tests/commands/test_check.py::test_check_cli_output 
tests/commands/test_lib.py::test_saving_deps 
tests/commands/test_ci.py::test_ci_empty 
tests/commands/test_init.py::test_init_default 
[gw0] [  1%] SKIPPED tests/commands/test_account_org_team.py::test_account_summary 
tests/commands/test_account_org_team.py::test_account_token 
[gw0] [  1%] SKIPPED tests/commands/test_account_org_team.py::test_account_token 
tests/commands/test_account_org_team.py::test_account_change_password 
[gw0] [  2%] SKIPPED tests/commands/test_account_org_team.py::test_account_change_password 
tests/commands/test_account_org_team.py::test_account_update 
[gw0] [  2%] SKIPPED tests/commands/test_account_org_team.py::test_account_update 
[gw3] [  3%] PASSED tests/commands/test_ci.py::test_ci_empty 
tests/commands/test_ci.py::test_ci_boards 
tests/commands/test_account_org_team.py::test_org_create 
[gw0] [  3%] SKIPPED tests/commands/test_account_org_team.py::test_org_create 
tests/commands/test_account_org_team.py::test_org_list 
[gw0] [  4%] SKIPPED tests/commands/test_account_org_team.py::test_org_list 
tests/commands/test_account_org_team.py::test_org_add_owner 
[gw0] [  4%] SKIPPED tests/commands/test_account_org_team.py::test_org_add_owner 
tests/commands/test_account_org_team.py::test_org_remove_owner 
[gw0] [  4%] SKIPPED tests/commands/test_account_org_team.py::test_org_remove_owner 
tests/commands/test_account_org_team.py::test_org_update 
[gw0] [  5%] SKIPPED tests/commands/test_account_org_team.py::test_org_update 
[gw4] [  5%] PASSED tests/commands/test_init.py::test_init_default 
tests/commands/test_init.py::test_init_ext_folder 
tests/commands/test_account_org_team.py::test_team_create 
[gw0] [  6%] SKIPPED tests/commands/test_account_org_team.py::test_team_create 
tests/commands/test_account_org_team.py::test_team_list 
[gw0] [  6%] SKIPPED tests/commands/test_account_org_team.py::test_team_list 
tests/commands/test_account_org_team.py::test_cleanup 
[gw0] [  7%] SKIPPED tests/commands/test_account_org_team.py::test_cleanup 
tests/commands/test_platform.py::test_search_json_output 
[gw4] [  7%] PASSED tests/commands/test_init.py::test_init_ext_folder 
tests/commands/test_init.py::test_init_duplicated_boards 
[gw4] [  8%] PASSED tests/commands/test_init.py::test_init_duplicated_boards 
tests/commands/test_init.py::test_init_ide_without_board 
[gw4] [  8%] PASSED tests/commands/test_init.py::test_init_ide_without_board 
tests/commands/test_init.py::test_init_ide_vscode 
[gw0] [  8%] PASSED tests/commands/test_platform.py::test_search_json_output 
tests/commands/test_platform.py::test_search_raw_output 
[gw3] [  9%] FAILED tests/commands/test_ci.py::test_ci_boards 

============================================== FAILURES ===============================================
___________________________________________ test_ci_boards ____________________________________________
[gw3] darwin -- Python 3.9.16 /Users/dada/Documents/projects/forks/platformio-core/.tox/py39/bin/python

clirunner = <click.testing.CliRunner object at 0x10da32790>
validate_cliresult = <function validate_cliresult.<locals>.decorator at 0x10da33940>

    def test_ci_boards(clirunner, validate_cliresult):
        result = clirunner.invoke(
            cmd_ci,
            [
                join("examples", "wiring-blink", "src", "main.cpp"),
                "-b",
                "uno",
                "-b",
                "leonardo",
            ],
        )
>       validate_cliresult(result)

tests/commands/test_ci.py:38: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

result = <Result SystemExit(2)>

    def decorator(result):
>       assert result.exit_code == 0, "{} => {}".format(result.exception, result.output)
E       AssertionError: 2 => Usage: ci [OPTIONS] [SRC]...
E         Try 'ci --help' for help.
E         
E         Error: Invalid value for '[SRC]...': Found invalid path: examples/wiring-blink/src/main.cpp
E         
E       assert 2 == 0
E        +  where 2 = <Result SystemExit(2)>.exit_code

tests/conftest.py:36: AssertionError
======================================= short test summary info =======================================
FAILED tests/commands/test_ci.py::test_ci_boards - AssertionError: 2 => Usage: ci [OPTIONS] [SRC]...
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! stopping after 1 failures !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
!!!!!!!!!!!!!!!!!!!!!!!! xdist.dsession.Interrupted: stopping after 1 failures !!!!!!!!!!!!!!!!!!!!!!!!
============================== 1 failed, 6 passed, 14 skipped in 13.56s ===============================
make: *** [test] Error 2
```

ivankravets, 2023-05-16:
You need to use `git clone --recursive`. See https://github.com/platformio/platformio-core/blob/develop/CONTRIBUTING.md

closed 2023-05-16 by ivankravets

MacDada, 2023-05-16:
> You need to use `git clone --recursive`

Oh, right, I missed that! :) Tests are working, thx.