platformio/platformio-core #4630: Error msg "File no found: Server error" when importing Arduino project

ThomasSlowhand, 2023-05-16:
>> Could not import Arduino project
Server error: {"type":"FileNotFoundError","args":[2,"Das System kann die angegebene Datei nicht finden"],"message":"[WinError 2] Das System kann die angegebene Datei nicht finden: ''C:\\\\Users\\\\ernst\\\\Documents\\\\PlatformIO\\\\Projects\\\\230516-065115-esp32-s3-devkitc-1'"}

### Configuration

Win 11:

Version: 1.78.2 (user setup)
Commit: b3e4e68a0bc097f0ae7907b217c1119af9e03435
Datum: 2023-05-10T14:39:26.248Z
Electron: 22.5.2
Chromium: 108.0.5359.215
Node.js: 16.17.1
V8: 10.8.168.25-electron.0
Betriebssystem: Windows_NT x64 10.0.22000
Sandkasten: Yes
### Description of problem

#### Steps to Reproduce

Open "Import Arduino" from Homepage
Select "ESP32-S3 DevKitC-1 as platform
Try to open the import folder "C:\Enwicklung\Arduino\factory"

[image]

ivankravets, 2023-05-16:
Do you use any russian espionage software, antivirus? In any case, create an empty PlatformIO project using the board and Arduino framework and just copy your Arduino files to the `src` folder.

ThomasSlowhand, 2023-05-16:
Hello Ivan,

 

Many thanks for reaching out, as you might know, anybody is using Russian E-spy software, or is being compromised with it. In fact, I have disabled Windows Defender, without any change. Although creating to complained folders by hand does not help. Please regards the curved bracket at the end of the error message line, which, in my opinion, has no place there. In the project explorer the by the error message complained folders (which I have created created manually) are shown without the leading number groups. And it seems to be from the threads regarding the same problem, that I am not alone with it.

 

Mit freundlichen Grüßen,

best regards

 

Thomas Krim