platformio/platformio-core #4655: error trying to flash ESP8266 via vbox HAOS

brokenhero000, 2023-06-10:
I tried to setup an ESP8266 in ESPHome, I flashed via virtualbox Home Assistant OS, and then I got an error that read I should report it to developers, so here it goes.

`INFO ESPHome 2023.5.5
INFO Reading configuration /config/esphome/esphome-web-8e15f8.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esphome-web-8e15f8 (board: esp01_1m; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/ESPAsyncTCP-esphome @ 1.2.3
INFO Installing esphome/ESPAsyncTCP-esphome @ 1.2.3
Error: Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/platformio/__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.9/dist-packages/platformio/cli.py", line 71, in invoke
    return super().invoke(ctx)
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/platformio/run/cli.py", line 147, in cli
    process_env(
  File "/usr/local/lib/python3.9/dist-packages/platformio/run/cli.py", line 204, in process_env
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
  File "/usr/local/lib/python3.9/dist-packages/platformio/run/processor.py", line 83, in process
    install_project_env_dependencies(
  File "/usr/local/lib/python3.9/dist-packages/platformio/package/commands/install.py", line 132, in install_project_env_dependencies
    _install_project_env_libraries(project_env, options),
  File "/usr/local/lib/python3.9/dist-packages/platformio/package/commands/install.py", line 247, in _install_project_env_libraries
    env_lm.install(
  File "/usr/local/lib/python3.9/dist-packages/platformio/package/manager/_install.py", line 48, in install
    pkg = self._install(spec, skip_dependencies=skip_dependencies, force=force)
  File "/usr/local/lib/python3.9/dist-packages/platformio/package/manager/_install.py", line 100, in _install
    pkg = self.install_from_registry(
  File "/usr/local/lib/python3.9/dist-packages/platformio/package/manager/_registry.py", line 29, in install_from_registry
    package = self.fetch_registry_package(spec)
  File "/usr/local/lib/python3.9/dist-packages/platformio/package/manager/_registry.py", line 93, in fetch_registry_package
    result = regclient.get_package(self.pkg_type, spec.owner, spec.name)
  File "/usr/local/lib/python3.9/dist-packages/platformio/registry/client.py", line 147, in get_package
    return self.fetch_json_data(
  File "/usr/local/lib/python3.9/dist-packages/platformio/http.py", line 158, in fetch_json_data
    return json.loads(result)
  File "/usr/lib/python3.9/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.9/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.9/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
`

and this is my .yaml config file 
`esphome:
  name: esphome-web-8e15f8
  friendly_name: ESPHome Web 8e15f8

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "f7gSxvxYoIu5ZzdyQVGTDU+Yrql69CMMoK8q832/Nd8="

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-8E15F8"
    password: "L0ED1ySDkhFB"

captive_portal:
    `

brokenhero000, 2023-06-13:
Interesting. I don't use proxy but I do own and use an antivirus called 360 total security, no way, can that really be the reason why my ESP8266s are struggling to connect? Note that I bought them from aliexpress for 2 euros or so, so they can be called fake to some extent

ivankravets, 2023-06-14:
Yes, it seems like "antivirus" spoils HTTP responses.

brokenhero000, 2023-06-14:
jeesus christ, so just to confirm, if I turn off my antivirus software I'll have better chances at connecting the ESP8266? hm that's really interesting tbh. I hope that works

ivankravets, 2023-06-14:
Today, most "antivirus" tools are espionage software. If you are on Windows, just use native Windows antivirus tools and don't trust others. See https://www.safetydetectives.com/blog/avast-scandal-why-we-stopped-recommending-avast-avg/

Also, the same story with russian software, such as Dr.web, Kaspersky. Try to avoid this espionage software.

brokenhero000, 2023-06-14:
Yes I'm on Windows. Well damn that's crazy, but honestly I tend to download a lot of cracked programs/apps from time to time, and this antivirus freeware called 360 total security protects me from viruses really well. Every time I download any game or a file, when I execute a program I get a prompt box telling me what kind of virus/threat was detected and asking if you wanna ignore it or delete it.

I'm just telling that a freeware antivirus software exists that is capable of that and I don't really trust the native Windows antivirus tools, they just tell you you've been infected lol, and they can't fix it unless you use an antivirus software, or at least that has been my experience for like half my life