platformio/platformio-core #4656: Download speeds suck in China. Here are some solutions.

CyberFoxHax, 2023-06-11:
reading this thread left me slightly miffed providing no solutions other than "got a new an shiny mirror" (that's dead)
https://github.com/platformio/platformio-core/issues/3835

ivankravets, 2023-06-11:
> I tried system variables, i tried set and setx, nothing worked. 

Why should it work? You set environment variable in your temporary terminal session. The VSCode works from the separated isolated process. You need to set the environment variables to global scope https://docs.oracle.com/en/database/oracle/machine-learning/oml4r/1.5.1/oread/creating-and-modifying-environment-variables-on-windows.html

> "got a new a new shiny mirror" (that's dead)

Duplicate of https://github.com/platformio/platformio-core/issues/4345 . Let's move to that issue.

closed 2023-06-11 by ivankravets

acmcee, 2023-08-27:
thanks，i can download platformio package successfully!

dennisdebel, 2025-11-02:
Having daily issues with platformio...it works for 10 hours and then you are stuck for two days....no matter proxy settings etc. As OP said, there is no feedback anywhere, not even in the inspector. What to do :(