platformio/platformio-core #4663: Problems compiling with esp32 after updating Platformio

qiudecai, 2023-06-15:
I get this error when compiling esp32 with Platformio, how to solve it?

Processing test_ESP32-wrover-kit_4M316k_ETH (platform: espressif32@1.12.2; board: esp-wrover-kit; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
 Compile time defines 
 File System:      SPIFFS
 PROGNAME:         ESP_mega_20230615_test_ESP32-wrover-kit_4M316k_ETH
 BUILD_PLATFORM:   Windows-10-10.0.19041-SP0
 GIT_HEAD:         pygit2_not_installed
 ------------------------------- 
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp-wrover-kit.html
PLATFORM: Espressif 32 (1.12.2) > Espressif ESP-WROVER-KIT
HARDWARE: ESP32 240MHz, 320KB RAM, 1.81MB Flash
DEBUG: Current (ftdi) On-board (ftdi) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.10004.210126 (1.0.4)
 - tool-esptoolpy @ 1.20600.0 (2.6.0)
 - tool-openocd-esp32 @ 1.1000.20190709 (10.0)
 - toolchain-xtensa32 @ 2.50200.97 (5.2.0)
Converting ESPEasy.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 69 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESPeasySerial @ 2.0.3+sha.737cfaf
|-- Adafruit ILI9341 @ 1.5.12
|-- Adafruit GFX Library @ 1.7.5
|-- TinyGSM @ 0.11.3
|-- Adafruit SGP30 Sensor @ 1.0.5
|-- Adafruit TCS34725 @ 1.0.1
|-- Adafruit TSL2591 Library @ 1.0.2
|-- AS_BH1750
|-- B-LUX-V30B
|-- ccronexpr
|-- ESP8266_SSD1306 @ 3.2.7
|-- WiFi @ 1.0
|-- HTTPClient @ 1.2
|-- Wire @ 1.0.1
|-- Adafruit Motor Shield V2 Library @ 1.0.4
|-- Adafruit MPR121 @ 1.0.0
|-- Adafruit NeoPixel @ 1.0.6
|-- Adafruit Unified Sensor @ 1.0.2
|-- AM2320
|-- ArduinoJson @ 6.11.0
|-- Blynk @ 0.6.1
|-- DLBus
|-- HLW8012 @ 1.1.1
|-- CHT16K33 LED Matrix Library @ 0.0.1
|-- SerialSensors @ 0.0.1
|-- LiquidCrystal_I2C
|-- NewPingESP8266
|-- PubSubClient @ 2.7
|-- MechInputs @ 0.0.1
|-- Regexp @ 0.1.0
|-- RN2xx3 Arduino Library @ 1.0.1
|-- SDM @ 2.0.0
|-- SparkFun APDS9960 RGB and Gesture Sensor @ 1.4.2
|-- SPI @ 1.0
|-- TinyGPSPlus @ 1.0.2
|-- SD(esp32) @ 1.0.5
|-- WebServer @ 1.0
|-- SPIFFS @ 1.0
|-- DNSServer @ 1.1.0
|-- ESPmDNS @ 1.0
|-- FS @ 1.0
|-- I2Cdevlib-Core
|-- ArduinoOTA @ 1.0
|-- ESP32HTTPUpdateServer
TypeError: can only concatenate deque (not "list") to deque:
  File "C:\Users\qdc\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 174:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\qdc\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\qdc\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\qdc\.platformio\platforms\espressif32@1.12.2\builder\main.py", line 213:
    target_elf = env.BuildProgram()
  File "C:\Users\qdc\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\qdc\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 61:
    env.ProcessProjectDeps()
  File "C:\Users\qdc\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\qdc\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 164:
    "$BUILD_SRC_DIR", "$PROJECT_SRC_DIR", env.get("SRC_FILTER")
  File "C:\Users\qdc\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\qdc\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 352:
    nodes = env.CollectBuildFiles(variant_dir, src_dir, src_filter)
  File "C:\Users\qdc\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\qdc\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 298:
    new_node = callback(new_node)
  File "E:\ESP8266\ESPEasy\kretzp_ESPEasy\RT_CJQ\tools\pio\generate-compiletime-defines.py", line 67:
    + [("SET_BUILD_GIT_HEAD", wrap_quotes(get_git_description()))],
====================================================== [FAILED] Took 7.80 seconds ======================================================
Environment                       Status    Duration
--------------------------------  --------  ------------
test_ESP32-wrover-kit_4M316k_ETH  FAILED    00:00:07.803
================================================ 1 failed, 0 succeeded in 00:00:07.803 ================================================

qiudecai, 2023-06-15:
Now back to Platformio 5.2.5 the problem persists

ivankravets, 2023-06-15:
This is not a problem with PIO Core. Please use the latest version of PIO Core.

1. "espressif32@1.12.2" - why do you use so outdated development platform? See https://github.com/platformio/platform-espressif32/releases
2. Please check this broken file generate-compiletime-defines.py