platformio/platformio-core #4667: `build_src_flags` impacts not only `src_dir` but also dependencies?

MacDada, 2023-06-18:
- [x] **PlatformIO Core**.

**PlatformIO Version** (`platformio --version`): `PlatformIO Core, version 6.1.7`

### Description of problem

According to [the docs](https://docs.platformio.org/en/latest/projectconf/sections/env/options/build/build_src_flags.html) `build_src_flags` _"will be applied **only for the project source files** in the [src_dir](https://docs.platformio.org/en/latest/projectconf/sections/platformio/options/directory/src_dir.html#projectconf-pio-src-dir) directory"_.

The problem is, it also applies to dependencies (external libraries) while… running tests :P

#### Steps to Reproduce

1. `platform: native`
2. `lib_deps = fabiobatsilva/ArduinoFake@0.3.1`
3. `build_src_flags = -Wpedantic`
4. `pio test`

### Actual Results

```
In file included from test/native/DnAppLogger/Decorator/test_SequenceLogger/SequenceLoggerTest.cpp:8:
In file included from lib/DnAppCommon/src/DnApp/Common/Strings.h:3:
In file included from arduino/cstdarg/cstdarg:13:
In file included from .pio/libdeps/native_arduino/ArduinoFake/src/Arduino.h:1:
In file included from .pio/libdeps/native_arduino/ArduinoFake/src/ArduinoFake.h:11:
.pio/libdeps/native_arduino/ArduinoFake/src/fakeit/fakeit.hpp:5749:27: warning: zero size arrays are an extension [-Wzero-length-array]
        char instanceArea[SIZE ? SIZE : 0];
                          ^~~~~~~~~~~~~~~
.pio/libdeps/native_arduino/ArduinoFake/src/fakeit/fakeit.hpp:6027:36: note: in instantiation of template class 'fakeit::FakeObject<FunctionFake>' requested here
        static_assert(sizeof(C) == sizeof(FakeObject<C, baseclasses...>), "This is a problem");
                                   ^
.pio/libdeps/native_arduino/ArduinoFake/src/fakeit/fakeit.hpp:7935:41: note: in instantiation of template class 'fakeit::DynamicProxy<FunctionFake>' requested here
        DynamicProxy<C, baseclasses...> _proxy;
                                        ^
```

### Expected Results

Clean test results, without the issues from external `ArduinoFake` library.

If I remove `build_src_flags = -Wpedantic`, the issues do not appear, so clearly `build_src_flags` impacts code outside of the `src` dir…

---

* a.) There is a bug in platformio core –> `build_src_flags` should not impact code from external libraries.
* b.) There is a bug in platformio docs -> it should be better explained what's actually going on.

ivankravets, 2023-06-19:
Could you try to run in verbose mode `-vvv`? Do you see `-Wpedantic` for dependencies? 

Please take a look at the chain, flag was applied for `SequenceLoggerTest.cpp:8:` where later that file included nested headers...

ivankravets, 2023-06-21:
Please reopen if you still need help

closed 2023-06-21 by ivankravets

omani, 2025-08-31:
I have to reopen this. build_src_flags definitely impacts dependencies. not on the first run, but on the subsequent runs. as soon as I run `pio run -t clean` the very first build (`pio run`) respects `build_src_flags` again. and any subsequent run does not.