platformio/platformio-core #4677: pio remote run -t upload not working

cyberman54, 2023-06-25:
I can't run firmware on my remote agent without using "-r" option, because firmware.bin is not found:

```
Environment    Status    Duration
-------------  --------  ------------
usb            SUCCESS   00:00:33.523
================================================ 1 succeeded in 00:00:33.523 ================================================ 
Uploading firmware remotely
Processing usb (framework: arduino; board: esp32dev; platform: espressif32@6.3.2)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option

warning: Calling missing SConscript without error is deprecated.
Transition by adding must_exist=False to SConscript calls.
Missing SConscript 'src/build.py'
File "/home/pi/.platformio/penv/lib/python3.9/site-packages/platformio/builder/main.py", line 168, in <module>
*** [upload] Source `.pio/build/usb/firmware.bin' not found, needed by target `upload'.
========================== [FAILED] Took 2.68 seconds ==========================
```

pio system info of local machine:

PlatformIO Core             6.1.7
Python                      3.8.10-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\aaa\.platformio
PlatformIO Core Executable  C:\Users\aaa\.platformio\penv\Scripts\platformio.exe
Python Executable           C:\Users\aaa\.platformio\penv\Scripts\python.exe
Global Libraries            0
Development Platforms       1
Tools & Toolchains          7

pio system info of remote agent machine:

PlatformIO Core             6.1.7
Python                      3.9.2-final.0
System Type                 linux_aarch64
Platform                    Linux-6.1.21-v8+-aarch64-with-glibc2.31
File System Encoding        utf-8
Locale Encoding             UTF-8
PlatformIO Core Directory   /home/pi/.platformio
PlatformIO Core Executable  /home/pi/.local/bin/platformio
Python Executable           /home/pi/.platformio/penv/bin/python
Global Libraries            0
Development Platforms       1
Tools & Toolchains          7

ivankravets, 2023-06-26:
Could you try to put shared files (extra scripts) to https://docs.platformio.org/en/latest/projectconf/sections/platformio/options/directory/shared_dir.html ?

cyberman54, 2023-06-26:
Thanks for this useful hint. I will try and report result here.

cyberman54, 2023-06-28:
I reinstalled platformio remote, following the instructions [here](
https://techoverflow.net/2021/10/05/how-to-install-platformio-remote-agent-on-raspberry-pi/) and started pio remote agent.
The agent seems to start successfully:
```
2023-06-28 13:26:19 [info] Connecting to PlatformIO Remote Development Cloud
2023-06-28 13:26:19 [info] Successfully connected
2023-06-28 13:26:19 [info] Authenticating
2023-06-28 13:26:20 [info] Successfully authorized
2023-06-28 13:26:36 [info] Remote command received: psync
2023-06-28 13:26:38 [info] Remote command received: run

```
If i now try to compile my project based on platformio-espressif32 framework, i get this error:

```
PLATFORM: Espressif 32 (6.3.2) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20009.0 (2.0.9)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5

**warning: Calling missing SConscript without error is deprecated.
Transition by adding must_exist=False to SConscript calls.
Missing SConscript '/home/pi/.platformio/packages/framework-arduinoespressif32/tools/platformio-build.py'
File "/home/pi/.platformio/platforms/espressif32/builder/frameworks/arduino.py", line 41, in <module>**
```

What can do to fix this?

/cc @ivan

cyberman54, 2023-06-28:
Checked the directory on the remote pi: indeed _platformio-build.py_ is missing.
It seems the installation is broken / incomplete.
And why are there .exe files on the linux based pi?

```
pi@prenzlpi:~/.platformio/packages/framework-arduinoespressif32/tools $ ls
gen_esp32part.exe  gen_insights_package.exe  gen_insights_package.py  ide-debug  sdk

```

ivankravets, 2023-06-28:
> **warning: Calling missing SConscript without error is deprecated.

Do you have any extra scripts in the `platformio.ini`?

cyberman54, 2023-06-28:
yes, there is a extra_scripts=pre:src/build.py in the platformio.ini

Meanwhile i worked around this by manually installing the framework arduino-esp32 files.

ivankravets, 2023-06-28:
It seems you missed my comment https://github.com/platformio/platformio-core/issues/4677#issuecomment-1606907360

closed 2023-06-28 by cyberman54

cyberman54, 2023-07-01:
@ivankravets Moving the pre: build script _AND_ all files that where referenced from this script from /src to /shared solved my problem. Thanks!