platformio/platformio-core #4682: cc1plus.exe fatal error

roberbike, 2023-07-01:
Hello.
I have this error compiling one proyect in windows 10.:

[image]

My platformio.ini has this flag that complile ok under linux :
build_flags =
${common.build_flags}
-D USER_SETUP_LOADED=1
-include /$PROJECT_LIBDEPS_DIR/$PIOENV/TFT_eSPI/User_Setups/Setup25_TTGO_T_Display.h

If I changue this line:
-include /$PROJECT_LIBDEPS_DIR/$PIOENV/TFT_eSPI/User_Setups/Setup25_TTGO_T_Display.h

by this other:
-include $PROJECT_LIBDEPS_DIR/$PIOENV/TFT_eSPI/User_Setups/Setup25_TTGO_T_Display.h

The proyect compile under windows 10 + vscode +platformio.
But not compile under linux. It is neccesary to put " / " to compile it.

There is an issue with platformio that cause this fault?

I asked here for this bug: https://community.platformio.org/t/cc1plus-exe-fatal-error/34551

Thanks.
Robert.

ivankravets, 2023-07-03:
Could you try a direct declaration using interpolation? See docs and examples here https://docs.platformio.org/en/latest/projectconf/interpolation.html

Try the next:
```ini
[env:myenv]
build_flags =
  -D USER_SETUP_LOADED=1
  -include ${platformio.libdeps_dir}/${this.__env__}/TFT_eSPI/User_Setups/Setup25_TTGO_T_Display.h    

```

roberbike, 2023-07-03:
This works under linux and windows:

 -include .pio/libdeps/${PIOENV}/TFT_eSPI/User_Setups/Setup25_TTGO_T_Display.h

ivankravets, 2023-07-03:
Sorry, I deleted my previous comment. This is actually a bug and we will fix it ASAP.

ivankravets, 2023-07-03:
Thanks for the report. Please re-test with `pio upgrade --dev`.

CapnBry, 2023-07-05:
I've tested with `PlatformIO Core, version 6.1.9a1` as 6.1.8 has broken our build system entirely and I still have an include error issue. Our build flags have a relative path in them but even when I add the `${PROJECT_SRC_DIR}/` to it to try to get an absolute path the build system seems to somehow still make an absolute path into the builder directory.
```
build_flags =
	${env_common_stm32.build_flags}
	${radio_2400.build_flags}
	${common_env_data.build_flags_tx}
	-flto
	-include target/FM30_TX.h
```

Build output
```
arm-none-eabi-gcc -o .pio\build\FM30_TX_via_DFU\FrameworkArduinoVariant\PeripheralPins.c.o -c -std=gnu11 -O2 -Wno-builtin-macro-redefined -Wall -flto -include C:\Users\bmayland\.platformio\platforms\ststm32\builder\${PROJECT_SRC_DIR}\target\FM30_TX.h 
```

Should I open a new issue for this or keep it here?

ivankravets, 2023-07-05:
Could you provide a simple project (zip archive attached here) to reproduce this issue?

CapnBry, 2023-07-05:
I apologize for my error, but I have misspoken in my previous message. The builder **does** properly support it if I supply an absolute path for the include file via a variable, but does not leave relative paths alone. It expands the relative path into the builder directory.

I do not have a simple project to supply. I can work around this in our project by supplying the full path the `-include` file but that's a real problem because we'd now have to go back and cut new version of all our supported older versions so the build works again.

ivankravets, 2023-07-05:
Please provide the 1 line from `platformio.ini` which works and another one that does not.

CapnBry, 2023-07-05:
working
```
build_flags =
	${env_common_stm32.build_flags}
	${radio_2400.build_flags}
	${common_env_data.build_flags_tx}
	-flto
	-include ${PROJECT_DIR}/include/target/FM30_TX.h
	-DUSBCON
	-DPIO_FRAMEWORK_ARDUINO_ENABLE_CDC
	-DHSE_VALUE=16000000U
	-DVECT_TAB_OFFSET=0x1000U
	-Wl,--defsym=FLASH_APP_OFFSET=0x1000
```

not working (what we had before
```
build_flags =
	${env_common_stm32.build_flags}
	${radio_2400.build_flags}
	${common_env_data.build_flags_tx}
	-flto
	-include target/FM30_TX.h
	-DUSBCON
	-DPIO_FRAMEWORK_ARDUINO_ENABLE_CDC
	-DHSE_VALUE=16000000U
	-DVECT_TAB_OFFSET=0x1000U
	-Wl,--defsym=FLASH_APP_OFFSET=0x1000

ivankravets, 2023-07-05:
You need
```
-include include/target/FM30_TX.h
```

The `cwd` is the project root dir.

CapnBry, 2023-07-05:
Oh I see! I appreciate that.

In 6.1.8 it worked though, and for years it has worked. Not to be too snarky because I appreciate your help, but 

> Process file as if #include "file" appeared as the first line of the primary source file.

This file is in the include path and therefore should have no issue being included, so long as an absolute path was not added to it. Is there any way this can not be processed and have a bogus path added to it?

ivankravets, 2023-07-05:
I think we can improve this behavior. I reopen this issue.

ivankravets, 2023-07-05:
Or, could I ask you to create a new issue and refer to this one? It's better for history.

closed 2023-07-05 by ivankravets

CapnBry, 2023-07-05:
Can do, will tag this issue in it too.

commit 01ab1fa4 (2023-07-03): Resolved a critical issue related to the usage of the ``-include`` flag // Resolve #4682