platformio/platformio-core #4683: include in build_flags expands path when it should not expand

CapnBry, 2023-07-05:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

### Configuration

**Operating system**:

Windows 10 Pro 22H2 19045.3086

**PlatformIO Version** (`platformio --version`):

PlatformIO Core, version 6.1.8
PlatformIO Core, version 6.1.9a1

### Description of problem

A continuation of #4682, when an environment's `build_flags` contains an `-include` item with a relative path and the file is in gcc's include path, the build fails as the builder has converted the relative path into an invalid absolute one. The difference between this and #4682 is that issue had a variable in it that needed to be expanded, and this is strictly a relative path to an item in the gcc include path.  The [documentation](https://docs.platformio.org/en/stable/projectconf/sections/env/options/build/build_flags.html) states that `-include`:

> Process file as if #include "file" appeared as the first line of the primary source file.

However, in 6.1.8 this has changed as the build process now replaces any `-include` with an absolute path, even when the path is not valid. In 6.1.7 and previous versions the following flags work `build_flags = -Iinclude -include XXX.h` if XXX.h is in `${PROJECT_DIR}/include`. The path to XXX.h gets expanded in 6.1.8 to the builder directory:

```
C:\Users\bmayland\.platformio\platforms\ststm32\builder\target\FM30_TX.h
```

Setting the `build_flags = -include "${PROJECT_DIR}/include/XXX.h` does not work in 6.1.8 as the variable is not expanded, and in 6.1.9a1 the variable is expanded (thanks to 01ab1fa4c08d0f5f854b84d6098a111760ebbe1b), but it is not backward compatible with the previous versions, or the documentation.

Our project supports building older versions as well, meaning we'd have to cut new releases for all our old versions just due to this incompatibility.

### Suggested Fix 

Perhaps the include can just not expand the path if the generated absolute path does not exist?

builder.py
```py
    # fix relative path for "-include"
    for i, f in enumerate(result.get("CCFLAGS", [])):
        if isinstance(f, tuple) and f[0] == "-include":
            rel_path = f[1].get_path()
            p = env.subst(rel_path)
            if os.path.exists(p):
                result["CCFLAGS"][i] = (f[0], os.path.abspath(p))
            else:
                # keep relative path
                result["CCFLAGS"][i] = (f[0], rel_path)
```

### Additional info

I'm sorry I do not have a simple project to reproduce this issue, as our project is somewhat large and I'd have to start something from scratch. This is the project:
https://github.com/ExpressLRS/ExpressLRS/

Jason2866, 2023-07-05:
@ivankravets  This is a major bug. It breaks project Tasmota completely and every other project that does something like this
```
build_flags                 = ${esp_defaults.build_flags}
                              -include "sdkconfig.h"
```

How can we force to use PIO Core v6.1.7 as fast fix?

fyi @barbudor @arendst

TD-er, 2023-07-05:
Yep, also broke ESPEasy builds....

> How can we force to use PIO Core v6.1.7 as fast fix?

You could perhaps enforce a specific version of Platformio in the requirements.txt file?

ivankravets, 2023-07-06:
Similar issue https://github.com/platformio/platformio-core/issues/4683

Sorry, the hot fix 6.1.9 is coming.

ivankravets, 2023-07-06:
Please re-test the latest development version (`pio upgrade --dev`). If it works, we will release 6.1.9. Thanks!

closed 2023-07-06 by ivankravets

hvrauhal, 2023-07-06:
Fix works for me in the [ELRS Configurator](https://github.com/ExpressLRS/ExpressLRS/)  with `6.1.9rc1`

TD-er, 2023-07-06:
Just wondering, how to define such an unreleased version in requirements.txt?
As that's how I handle Python lib versions in my build env.

ivankravets, 2023-07-06:
Just pass 
```
platformio==https://github.com/platformio/platformio-core/archive/refs/heads/develop.zip
```

TD-er, 2023-07-06:
Apparently not, see: https://github.com/letscontrolit/ESPEasy/actions/runs/5474537142/jobs/9969453955#step:7:176

ivankravets, 2023-07-06:
We have just pushed a new solution to this issue. Could you restart the build?

ivankravets, 2023-07-06:
Sorry, my typo. See syntax https://pip.pypa.io/en/stable/reference/requirements-file-format/

This should work:

```
platformio @ https://github.com/platformio/platformio-core/archive/refs/heads/develop.zip
```

TD-er, 2023-07-06:
As the [first env](https://github.com/letscontrolit/ESPEasy/actions/runs/5474665105) completed already, I think it did indeed fix the builds again.

Jason2866, 2023-07-06:
Can confirm. Latest development version does work.

ivankravets, 2023-07-06:
Big THANKS for the confirmation. 6.1.9 is coming!

ivankravets, 2023-07-06:
6.1.9 is out! 

https://github.com/platformio/platformio-core/releases/tag/v6.1.9

CapnBry, 2023-07-06:
Yup I woke up and installed the dev version (which ended up being 6.1.10) which worked, rolled back to the release 6.1.9 (which was released 8 minutes earlier) and it also works again. Thank you!

commit 109c537d (2023-07-06): Fixed handling of  ``-include`` flag // Resolve #4683
commit 9585e2a3 (2023-07-06): Fixed handling of  ``-include`` flag // Resolve #4683