platformio/platformio-core #4684: Constant include errors with ClangD and generated compile_commands.json

jiriks74, 2023-07-05:
What kind of issue is this?
### Configuration

**Operating system**: KDE Neon 5.27

**PlatformIO Version**: 6.1.7

### Description of problem
ClangD includes either system header files or doesn't see them at all.

If you include `Arduino.h` you'll get this error if you don't have `gcc-multilib` installed on your system:
`clang: 'bits/libc-header-start.h' file not found [pp_file_not_found]`.

If you do have `gcc-multilib` on your system then you'll get this error instead:
`clang: In included file: 'avr/pgmspace.h' file not found [pp_file_not_found]`.

When looking into the `Arduino.h` file I can see that eg. the `bits/libc-header-start.h` is from the system header files instead of the Arduino ones:

<details>
<summary>Image</summary>

[image]

</details>

#### Steps to Reproduce

1. `pio project init --board uno`
2. `touch src/main.cpp`
3. Open `main.cpp`
4. Add `#include <Arduino.h>` and see the error
5. `pio run -t compiledb`
6. If present from the start create some header file in the `include` directory and do the same as in `main.cpp`

### Expected Results
ClangD correctly solving header paths so the code completion works

### If problems with PlatformIO Build System:

<details>
<summary><b>The content of <code>platformio.ini</code>:</b></summary>

```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:uno]
platform = atmelavr
board = uno
framework = arduino
```

</details>

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>
```

### Additional info
While writing this issue I figured out that it works as expected *if* you first run `pio project init --board uno && pio run -t compiledb` and then create the `main.cpp` file with `#include <Arduino.h>`. If you do it the other way around or you run `pio run -t compiledb` after creating the `main.cpp` file you'll get these errors.

ivankravets, 2023-07-06:
This is a bug and fixed it. Please re-test with `pio upgrade --dev`.

closed 2023-07-06 by ivankravets

jiriks74, 2023-07-06:
Thanks @ivankravets it works <3

ubaldot, 2024-01-06:
This issue `In included file: 'avr/pgmspace.h' file not found [pp_file_not_found]` seems to be back. 
I am running `PlatformIO Core, version 6.1.12b2`, this is my `.ccls` file:
```
clang

%c -std=gnu11 -fno-fat-lto-objects -mmcu=atmega328p -Os -Wall -ffunction-sections -fdata-sections -flto
%cpp -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -mmcu=atmega328p -Os -Wall -ffunction-sections -fdata-sections -flto

-I/Users/ubaldot/Documents/embedded/ArduinoFreeRTOS/SerialQueue/include
-I/Users/ubaldot/Documents/embedded/ArduinoFreeRTOS/SerialQueue/src
-I/Users/ubaldot/.platformio/packages/framework-arduino-avr/cores/arduino
-I/Users/ubaldot/.platformio/packages/framework-arduino-avr/variants/standard
-I/Users/ubaldot/.platformio/packages/framework-arduino-avr/libraries/EEPROM/src
-I/Users/ubaldot/.platformio/packages/framework-arduino-avr/libraries/HID/src
-I/Users/ubaldot/.platformio/packages/framework-arduino-avr/libraries/SPI/src
-I/Users/ubaldot/.platformio/packages/framework-arduino-avr/libraries/SoftwareSerial/src
-I/Users/ubaldot/.platformio/packages/framework-arduino-avr/libraries/Wire/src
-I/Users/ubaldot/.platformio/packages/toolchain-atmelavr/lib/gcc/avr/7.3.0/include
-I/Users/ubaldot/.platformio/packages/toolchain-atmelavr/lib/gcc/avr/7.3.0/include-fixed
-I/Users/ubaldot/.platformio/packages/toolchain-atmelavr/avr/include

-DPLATFORMIO=60112
-DARDUINO_AVR_UNO
-DF_CPU=16000000L
-DARDUINO_ARCH_AVR
-DARDUINO=10808
-D__AVR_ATmega328P__

```

I am using MacVim, using `clangd` as LSP server.

commit 837ea85c (2023-07-06): Resolved an issue that resulted in unresolved absolute toolchain paths when generating the "compile_commands.json" // Resolve #4684