platformio/platformio-core #4688: cppcheck doesn't seem to be able to handle __has_include() properly

ssilverman, 2023-07-11:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

### Configuration

**Operating system**: macOS 13.4.1

**PlatformIO Version** (`platformio --version`): 6.1.9

Building for a Teensy 4.1 and C++17. Added this line to _platformio.ini_:
```ini
build_unflags = -std=gnu++14 -std=c++14
```

### Description of problem

`cppcheck` doesn't seem to be able to handle `__has_include()` properly. Is it possible it's just not working with angle-bracket includes?

#### Steps to Reproduce

1. Run "Inspect" tool on any C++17 project.

### Actual Results

"failed to evaluate #if condition" errors that look like this:
```
Error: Found a breaking defect 'failed to evaluate #if condition' in ~/.platformio/packages/toolchain-gccarmnoneeabi-teensy/arm-none-eabi/include/c++/11.3.1/arm-none-eabi/bits/c++config.h:759
```

Code at line 759:
```c++
#if __has_include(<pstl/pstl_config.h>)
```

When I comment that out, it fails in other places. For example, line 37 in _~/.platformio/packages/toolchain-gccarmnoneeabi-teensy/arm-none-eabi/include/c++/11.3.1/ext/atomicity.h_:
```c++
#if __has_include(<sys/single_threaded.h>)
```

### Expected Results

No "failed to evaluate #if condition" errors.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env]
build_flags = -Wall -g -Og -fno-single-precision-constant
build_unflags = -std=gnu++14 -std=c++14
monitor_speed = 115200

[env:teensy41]
platform = teensy
board = teensy41
framework = arduino
```

**Source file to reproduce issue:**: N/A

### Additional info

ssilverman, 2023-07-11:
I wasn't sure if this is the correct spot, or if this issue needs to be in _platformio-vscode-ide_.

valeros, 2023-07-12:
Hi @ssilverman, indeed Cppcheck's internal parser is quite limited and can fail to evaluate tricky code in some custom embedded packages. That's why we have the [check_skip_packages](https://docs.platformio.org/en/latest/projectconf/sections/env/options/check/check_skip_packages.html) option for such cases.

closed 2023-07-12 by valeros