platformio/platformio-core #4695: prject.checksum added in wrong folder

jonnytest1, 2023-07-18:
What kind of issue is this?

- [x]  **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`):

### Description of problem
when using a build_dir with ${UNIX_TIME} a folder named "${UNIX_TIME}" is created with the project.checksum
instead it should use the same folder as the rest of the build

#### Steps to Reproduce

1. create a platformio.ini with UNIX_TIME for build dir
```ini
[platformio]
build_dir = ${platformio.workspace_dir}\build\${UNIX_TIME}
```
2. build

### Actual Results

project checksum is generated into a different folder that doenst take the variable into account correctly
[image]

### Expected Results
project.checksum is added in the interpolated folder 

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[platformio]
build_dir = ${platformio.workspace_dir}\build\${UNIX_TIME}
```

jonnytest1, 2023-07-18:
🤔 for now it doesnt seem like a huge problem , upload still worked at least

salih-06, 2023-07-23:
Arduino için yazmış olduğum kod Arduino IDE üzerinde sorunsuz çalışırken bu IDE de loop bloğu içinde fonksiyon çağırdığımda
hatta alıyorum lütfen sorun çözülsün