platformio/platformio-core #4698: clang-tidy inconsistent outputs between local and ci

aidiaz, 2023-07-25:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to the
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues) repository

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`):
PlatformIO Core, version 6.1.9
### Description of problem
pio check gets different outputs in local respect to CI implementation

#### Steps to Reproduce

1. run check on local project
2. run check on CI with github action
3. compare outputs

### Actual Results
```
Checking pico32 > clangtidy (platform: espressif32; board: pico32; framework: espidf)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
main/main.c:1: [medium:warning] invalid case style for variable 'prevTime'  [readability-identifier-naming]
========================================================================== [PASSED] Took 0.04 seconds ==========================================================================

Component     HIGH    MEDIUM    LOW
-----------  ------  --------  -----
main           0        1        0

Total          0        1        0

Environment    Tool       Status    Duration
-------------  ---------  --------  ------------
pico32         clangtidy  PASSED    00:00:00.040
========================================================================== 1 succeeded in 00:00:00.040 ==========================================================================
 *  Terminal will be reused by tasks, press any key to close it. 

```
### Expected Results
```
pio check --fail-on-defect=high --fail-on-defect=medium
********************************************************************************
If you like PlatformIO, please:
- star it on GitHub > https://github.com/platformio/platformio-core
- follow us on LinkedIn to stay up-to-date on the latest project news > https://www.linkedin.com/company/platformio/
[... 163 lines cut ...]

on:
  pull_request:
   types: [opened, reopened, edited, synchronize]
jobs:
  lint:
      runs-on: ubuntu-latest
      steps:
        - name: Checkout repository
          uses: actions/checkout@v3
        - name: Cache pio and pip
          uses: actions/cache@v3
          with:
            path: |
              ~/.cache/pip
              ~/.platformio/.cache
            key: ${{ runner.os }}-pio
        - uses: actions/setup-python@v4
          with:
            python-version: '3.9'
        - name: Install PlatformIO Core
          run: pip install --upgrade platformio
        - name: PIO Lint
          run: pio check --fail-on-defect=high --fail-on-defect=medium
```

valeros, 2023-07-28:
Hi @aidiaz, indeed I can reproduce the issue, but I'm curious how you managed to run clang-tidy locally. I tried to run your code snippet and besides some `clang-diagnostic-error` regarding undeclared type, I also got only one `readability-identifier-naming` issue:

```
warning: invalid case style for variable 'prevTime' [readability-identifier-naming]
uint8_t prevTime = 0;
        ^~~~~~~~
        prev_time

```

aidiaz, 2023-07-30:
I have installed clang-tidy via brew and run it from my console

valeros, 2023-07-31:
> I have installed clang-tidy via brew and run it from my console

I'm interested in the precise command you're invoking. Also, please share here a complete source file with all include directives.

aidiaz, 2023-07-31:
In my local I run 
```
clang-tidy main/* --config-file='.clang-tidy'
```

valeros, 2023-08-02:
> In my local I run

Please also share a complete source file with all include directives.

valeros, 2023-09-07:
@aidiaz with your code above everything works as expected even locally:
[image]

Although I can reproduce the issue if I remove the include line `#include "stdint.h"`, but the behavior is similar to bare clang-tidy output with one exception that PlatformIO omits Clang's diagnostic errors.

[image]

aidiaz, 2023-09-07:
This is really weird, with the code above my local returns:
[image]

valeros, 2023-09-07:
@aidiaz Do you mind sharing a minimal but fully configured PlatformIO project as a zip archive?

aidiaz, 2023-09-07:
Sure! here is the project Im testing with
[clang-tidy-test.zip](https://github.com/platformio/platformio-core/files/12551848/clang-tidy-test.zip)

valeros, 2023-09-08:
Thanks! The project configuration shown in the issue description contains the `check_skip_packages = yes` option while the project you're testing with doesn't have this line. Is it intentional? Does it work if you enable it?

aidiaz, 2023-09-12:
The `chech_skip_packages = yes` option seems to change nothing in this project. 
But, noticed that when running a "Verbose Check"  it outputs `stdint.h not found`.
[image]

valeros, 2023-09-14:
That's indeed strange, your exact project example just with enabled `check_skip_packages = yes` generates proper results on my machine:
[image]
which makes me think that clang-tidy's behavior may somewhat depend on the system environment it's running in. I'll get back to you once I have new info.

closed 2024-02-08 by ivankravets