platformio/platformio-core #4706: Platformio is always checking for an internet connection when a task is executed offline

gdamino, 2023-08-04:
### Configuration

**Operating system**: MacOS Ventura 13.4

**PlatformIO Version** (`platformio --version`): 6.1.9

### Description of problem
Using Platformio as Extension of Visual Studio Code, programming with ESP32 Idf framework:
When internet connection is down (and is always down in our case because we are connecting to the Wifi Hotspot created by the product) when you start a task like compile, upload or start serial monitor the command is searching for a internet connection, So: 
`platformio device monitor` will stay locked for minutes or until I change wifi network to one with internet available.

Note: this issue is appaered with last builds of Platformio, I don't know what exactly, but I never noticed it in years before.

#### Steps to Reproduce
Stay offline.
1. Do something like upload firmware or start serial monitor

### Actual Results
Task delay to an undertemined time before start, like 30 seconds, 1 minute.

### Expected Results
Please make possible to use tools and script that can be used offline to be used offline. Why a `platformio device monitor ` needs to connect to internet every time it runs??

ivankravets, 2023-08-04:
Thanks for the report! 

Could you re-test with the latest development version? Open terminal and type:
```
pio upgrade --dev
```

Does it work now?

closed 2023-08-04 by ivankravets

gdamino, 2023-08-05:
Yes, with last dev version the issue disappaered. I will update this thread if I will notice it again. Thanks

ivankravets, 2023-08-05:
Thanks for the updates. We bump a new release next week.

commit 48a823d3 (2023-08-04): Do not check internet connection on the postponed telemetry // Resolve #4706