platformio/platformio-core #4710: Add  UNITY_INCLUDE_CONFIG_H and path to it in c_cpp_properties.json

dzid26, 2023-08-09:
What kind of issue is this?

- [x] **Feature Request**.
------------------------------------------------------------------

**PlatformIO Version** (`platformio --version`):
PlatformIO Core, version 6.1.9

### Description of the problem

Intellisense doesn't find some functions in my unit test because `UNITY_INCLUDE_CONFIG_H`  and `unity_config.h` include path are not defined in `c_cpp_properties.json`. 
[image]

`.pio/libdeps/ServoCAN/Unity/src",` is already in `c_cpp_properties.json`...

Note, I use `unity_config.h` to define some common top-level features like semihosting and: 
```
#define UNITY_INCLUDE_PRINT_FORMATTED

#ifdef USE_STDPERIPH_DRIVER
#include "stm32f10x.h"
#elif HAL_MODULE_ENABLED
#include "stm32f1xx_hal.h"
#endif
```

### Expected Results

To have "UNITY_INCLUDE_CONFIG_H", and                     
and path to `unity_config.h` in `c_cpp_properties.json` so that dependencies are resolved properly.  

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

dzid26, 2023-08-10:
Hmm, I noticed `UNITY_INCLUDE_CONFIG_H` appears in `c_cpp_properties.json` when unit testing env is selected. I guess I would prefer for it to be added if a single test is detected in the workspace.

Conversely `.pio/libdeps/Release/Unity/src"` is added to `c_cpp_properties.json` for any env, as long as the Unity directory exists.

dzid26, 2023-08-10:
Yes, I go through all my envs using VScode sidebar unittester. It pulls unity sources into here:
[image]
That's works, although, for the release env, I need to fake that I am doing the test in order to pull Unity and for the path to `libdeps/xxx_release/Unity/src` to appear in `c_cpp_properties.json` to appear. 

But that's only this one path define. Others, i.e. `UNITY_INCLUDE_CONFIG_H` and path to `unity_config.h` don't appear automatically for the release. 
I don't like to switch away from the release though, because then Upload button becomes useless. 
 
Btw, I can't get `UNITY_INCLUDE_CONFIG_H`  to appear `c_cpp_properties.json` anymore even by switching env to unit test one.  

I need to figure out what triggers this stuff:
https://github.com/platformio/platformio-core/blob/65397fe0591eb3d37056effcb015f5c1893a7206/platformio/test/runners/unity.py#L215-L226

ivankravets, 2024-02-09:
Could you provide a simple project (as a ZIP archive attached to this issue) to reproduce this issue?