platformio/platformio-core #4711: Create Compilation Database on every build

dzid26, 2023-08-10:
- [x] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

I would like `compile_commands.json` generation 
https://github.com/platformio/platformio-core/blob/65397fe0591eb3d37056effcb015f5c1893a7206/platformio/builder/main.py#L151
to be triggered by adding a deliberate flag in platformio.ini. 

```
[env]
targets = upload, compiled
``` 
is not good because it changes the VSCode GUI icons' action.

dzid26, 2024-01-26:
I tried to call this from in `extra_scripts` (post):
```
import subprocess
projectTag = (
    subprocess.run(['pio', 'run','-t','compiledb'])
)
```
but the build went into infinite loop and started eating up RAM.

ivankravets, 2024-01-26:
Do you run that part of the code under the guard?

```
if "compiledb" not in COMMAND_LINE_TARGETS: 
    subprocess.run(['pio', 'run','-t','compiledb'])
```

dzid26, 2024-01-26:
Ah, that works great.

```
[env]
extra_scripts = post:generate_compilation_database.py #for the sonarlint
```
```
import subprocess
if "compiledb" not in COMMAND_LINE_TARGETS: #avoids infinite recursion
    subprocess.run(['pio', 'run','-t','compiledb'])
```

ivankravets, 2024-01-27:
I think this is a great solution. I'm going to close this issue.

Happy coding with PlatformIO! 🚀

closed 2024-01-27 by ivankravets