platformio/platformio-core #4713: [Feature Request] install library as "editable" (simillar to pip install -e)

cr1tbit, 2023-08-11:
### Configuration

**Operating system**: Linux (I use Arch btw.)

**PlatformIO Version** (`platformio --version`): `PlatformIO Core, version 6.1.11a1`

### Description of problem
When developing custom platformio library I need fast development cycle. This means that I can change a line in my library and then compile the code to see the result.

For now, reading documentation, i came up with 3 solutions:

#### 1. Pushing my library changes to a git repo, and define the dependency as `https://github.com/[...]`

Not ideal, every change needs to be pushed

#### 2. Defining the dependency as present on local filesystem, using `file:///[PATH]`

Better, but the problem is that when building, the library is copied to the `.pio/libdeps` folder, so after doing each change i need to clear `libdeps` folder (even cleaning the project perserves `libdeps`)

#### 3. Symlinking the local project's location to the `lib` folder

This works almost perfectly, with one quirk that the buildsystem seems to omit finding dependencies from `lib/[LIB_NAME]/library.json`. After adding missing dependencies to the `platformio.ini` manually, the project compiles fine.

Also, this workflow won't work on Windows probably.

### Potential solutions

The soultion i have in mind is providing something like `pip install -e` in python, where the library can be installed as "editable", meaning during buildsystem, the files are included directly from the library's directory, instead of being copied to `.pio/libdeps`.

Documentation: https://pip.pypa.io/en/stable/topics/local-project-installs/

If that solution isn't feasible, just parsing `lib/[LIB_NAME]/library.json` for the dependencies would be sufficient.

ivankravets, 2023-08-14:
You need to use `symlink://...`. See https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_install.html#local-folder

closed 2023-08-14 by ivankravets

jmnc2, 2024-06-18:
> You need to use `symlink://...`. See https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_install.html#local-folder

When I symlink with VS Code, the links in the PROBLEMS tab add the working dir to the start of the full path and then offer to create a new file for you :(

I am also using arch/Manjaro. Compiling etc works fine.

As below:
[{
	"resource": "/home/jmnc2/doc/src/Arduino/homeauto_network/home/jmnc2/doc/src/Arduino/libraries/defs/src/defs.c",
	"owner": "cpp",
	"severity": 8,
	"message": "expected '=', ',', ';', 'asm' or '__attribute__' at end of input",
	"startLineNumber": 31,
	"startColumn": 1,
	"endLineNumber": 31,
	"endColumn": 1
}]

ivankravets, 2024-06-18:
@jmnc2 , could you file a new issue? Thanks 🙏