platformio/platformio-core #4714: Arduino ESP32 Board Please Add

eddyhodak, 2023-08-14:
What kind of issue is this?

- [X ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

The New Arduino ESP32 Board would be a great addition.

maxgerhardt, 2023-08-14:
What? This was already handled in 

https://github.com/platformio/platform-espressif32/issues/1162

ivankravets, 2023-08-14:
Duplicate of https://github.com/platformio/platform-espressif32/issues/1162

closed 2023-08-14 by ivankravets

eddyhodak, 2023-08-14:
Cheers, Ivan

eH!
Enjoying This Day

 
 

Closed #4714 as completed.

eddyhodak, 2023-08-14:
Cheers, Maximilian

eH!
Enjoying This Day

 
 

platformio/platform-espressif32#1162