platformio/platformio-core #4716: "build_dir" seemed not working.

swortain, 2023-08-15:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to the
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues) repository

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

Hi, I'm not sure if there is sth wrong with my setting. but "build_dir" seemed not work no matter what I set in "platformio.ini" file
[image]
And when I dump the env in script. it is shown like this.
[image]

maxgerhardt, 2023-08-15:
You didn't read the documentation well. `build_dir` only works in the special `[platformio]` section, you put it in a regular `[env:xxx]` section.

https://docs.platformio.org/en/latest/projectconf/sections/platformio/index.html

ivankravets, 2023-08-15:
Thanks, Max! 🙏

closed 2023-08-15 by ivankravets

swortain, 2023-08-15:
got it. thanks, max!

> You didn't read the documentation well. `build_dir` only works in the special `[platformio]` section, you put it in a regular `[env:xxx]` section.
> 
> https://docs.platformio.org/en/latest/projectconf/sections/platformio/index.html
> 
> ```ini
> [platformio]
> build_dir = ...
> 
> [env:esp32-...]
> ...
> ```