platformio/platformio-core #4718: CMakeLists.txt and CMakeListsPrivate.txt no longer regenerated upon Re-Init (neither CLI nor CLion)

divStar, 2023-08-15:
### Configuration

**Operating system**: MacOS Ventura 13.5 (M1)

**PlatformIO Version** (broken on): 6.1.10
**PlatformIO Version** (working on): 6.1.9

### Description of problem
When using the following CLI command: `pio project init --board zero --ide clion`, I'd expect the `CMakeLists.txt` and `CMakeListsPrivate.txt` to be generated (and probably re-generated, if they already exist and differ e.g. due to additional flags in the `platformio.ini`).

In version `6.1.9` (and `6.1.9_1`, from what I have tried) of PlatformIO Core, at least the generation of these files works (haven't tried re-generating them). In version `6.1.10` it does not.

#### Steps to Reproduce
1. Create an empty PlatformIO project or use existing one.
2. Ensure `platformio.ini` exists and contains the `zeroUSB` environment (I believe I read somewhere, that nearly all environments cause `CMakeLists*.txt` generation, but **not all** of them).
3. Ensure no `CMakeLists.txt` and `CMakeListsPrivate.txt` exists in the project's root folder.
4. Use `pio project init --board zero --ide clion` command to initialize project.

### Actual Results
On PlatformIO `6.1.10`, I receive the following output:
```shell
Resolving zero dependencies...
Already up-to-date.
Updating metadata for the clion IDE...
Project has been successfully updated!
```
However: **no `CMakeLists.txt` nor `CMakeListsPrivate.txt` have been created**.

The output on `6.1.9` and `6.1.9_1` look essentially the same (expect for warnings of the version being obsolete), but both files are generated.

### Expected Results
As with version `6.1.9`, I'd expect `pio project init --board zero --ide clion` to generate the `CMakeLists.txt` and `CMakeListsPrivate.txt`.

### `platformio.ini`
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
description = HardwareSensors for the AhmsVille Dial2
default_envs = zeroUSB

[env:zeroUSB]
platform = atmelsam
board = zeroUSB
framework = arduino
upload_speed = 115200
lib_ldf_mode = chain+
lib_deps = 
	voronii/Dial2AbstractSensors@^1.0.1
	fastled/FastLED@^3.6.0
	bblanchon/ArduinoJson@^6.21.2
	thijse/ArduinoLog@^1.1.1
	arduino-libraries/RTCZero@^1.6.0
	ivanseidel/LinkedList@0.0.0-alpha+sha.dac3874d28
	paulstoffregen/CapacitiveSensor@^0.5.1
	denyssene/SimpleKalmanFilter@^0.1.0
build_flags = -std=gnu++17
build_unflags = -std=gnu++11
build_src_flags = -Wall -Wextra -Wpedantic
build_src_filter = 
	+<**/*.h>
	+<**/*.c>
	+<**/*.cpp>
```

### Additional info
Since in version `6.1.9` the generation of `CMakeLists.txt` and `CMakeListsPrivate.txt` works just as expected, yet does not work with version `6.1.10`, I am currently staying on `6.1.9`. I am unsure whether there was some migration to do or if perhaps some flags are outdated or something - if so, please tell me and I will give it a try.

Lastly: if it's necessary, I can provide the whole Git repository (`Dial2HardwareSensors` library) for testing purposes. It really is easy to reproduce.

divStar, 2023-08-16:
> @divStar Could you try updating Clion and PIO plugin as well ?

I will try that, though I think I am on the most recent version of the CLion plugin.

Do you think, that it might be for a reason, that the CMakeLists files aren't generated anymore and that the PIO plugin for CLion takes care of it by itself in their most recent versions?

I'm asking, because in version 6.1.9 the files could be generated from the CLI (without CLion completely).

ivankravets, 2023-08-16:
See https://blog.jetbrains.com/clion/2023/07/clion-2023-2-released/

PlatformIO plugin for CLion is fully independent of CMake.

closed 2023-08-16 by ivankravets

divStar, 2023-08-16:
@ivankravets - thank you, I updated PlatformIO to 6.1.10 and CLion to 2023.2 and it's working. However: the syntax highlighting (especially on missing or wrong includes etc.) is not showing. Is that a known issue and/or is there a workaround? Or is the workaround to use `Static Code Analysis` and decrypt the CLI output rather than seeing it in code in-place?

ivankravets, 2023-08-17:
ping @elmot

elmot, 2023-08-20:
Sounds like a good feature request. Please file a ticket at our [YouTrack](https://youtrack.jetbrains.com/newIssue?project=CPP&c=Subsystem%20Embedded&c=tag%20platformio) and provide a test project.
Better if the project is small, and a number  with reasonable diagnostics is generated while running the code analysis.