platformio/platformio-core #4728: VCSBaseException: VCS: Could not process command ['git', 'clone', '--recursive', 'https://github.com/earlephilhower/arduino-pico.git', 'C:\\Users\\<User>\\.platformio\\.cache\\tmp\\pkg-installing-yg5f4c_c']

joralllc, 2023-08-22:
### Configuration

**Operating system**: Win10 Pro

**PlatformIO Version** (`platformio --version`): 3.3.1

### Description of problem

> VCSBaseException: VCS: Could not process command ['git', 'clone', '--recursive', 'https://github.com/earlephilhower/arduino-pico.git', 'C:\\Users\\<User>\\.platformio\\.cache\\tmp\\pkg-installing-yg5f4c_c']
Thrown whenever PIO tries to initialize upon installation, reloading VSCode, or using PIO build

#### Steps to Reproduce

1. Install VSCode, Git, Python, PIO (latest versions, tried reinstalling all)
2. Let PIO try to set itself up for project involving RPi Pico
3. Wait

### Actual Results
Refer to Description of Problem

### Expected Results
Functional software

### Additional info
I have no experience with this software; I was told to use it by a coworker for a project they developed themself.

ivankravets, 2023-08-24:
You can use https://github.com/earlephilhower/arduino-pico.git directly. This is an related project.

closed 2023-08-24 by ivankravets

maxgerhardt, 2023-08-24:
FYI @joralllc likely didn't follow https://arduino-pico.readthedocs.io/en/latest/platformio.html#important-steps-for-windows-users-before-installing.

TheNubOfLegend, 2023-08-24:
@maxgerhardt was correct; my coworker's instructions only told me to install the VSCode extension and nothing else.  The problem is resolved now.