platformio/platformio-core #4730: .ino-files in subfolders are not converted to cpp

bimac, 2023-08-27:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Fedora 38

**PlatformIO Version**: 6.1.11a2

### Description of problem

PIO does not convert .ino files in subfolders when including the latter with build_src_filter

#### Steps to Reproduce

Create a project containing the following and the platformio.ini below.

├── src
│   ├── v1
│   │   ├── some_lib.cpp
│   │   ├── some_lib.h
│   │   └── v1.ino
│   ├── v2
│   │   ├── some_lib.cpp
│   │   ├── some_lib.h
│   │   └── v2.ino
│   ├── v3
│   │   ├── some_lib.cpp
│   │   ├── some_lib.h
│   │   └── v3.ino

### Actual Results

PIO does not convert .ino to .cpp, compiler only compiles some_lib, linker complains:
```
Linking .pio/build/v1/firmware.elf
.pio/build/v1/libFrameworkArduino.a(main.cpp.o): In function `main':
main.cpp:(.text.startup.main+0x22): undefined reference to `setup'
main.cpp:(.text.startup.main+0x26): undefined reference to `loop'
collect2: error: ld returned 1 exit status
*** [.pio/build/v1/firmware.elf] Error 1
```

It works fine when manually renaming the .ino files to .cpp

### Expected Results

PIO should find and convert .ino in subfolders

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env]
framework = arduino

[env:v1]
platform = atmelsam
board = sparkfun_samd21_mini_usb
build_src_filter =
	-<*>
	+<v1>

[env:v2]
platform = teensy
board = teensy31
build_src_filter =
	-<*>
	+<v2>

[env:v3]
platform = teensy
board = teensy40
build_src_filter =
	-<*>
	+<v3>
```

**Source file to reproduce issue:**
```cpp
#include "some_lib.h"

void setup() {

}

void loop() {

}
```

### Additional info
See https://community.platformio.org/t/multiple-src-folders-containing-ino-files

PBahner, 2023-08-28:
Hi,

The problem is that only .ino files in the **project src-dir** are converted to .cpp.

I am not familiar with pio source code but I researched the code a little bit because I have the same problem with my project.

### 1. src_filter: 

src_filter is only searching for the following file types: `"h", "hpp", "S", "spp", "SPP", "sx", "s", "asm", "ASM", "c", "cc", "cpp", "cxx", "c++"`
This may be no problem when the .ino files are first converted to .cpp??

```python
for ext in piobuild.SRC_BUILD_EXT + piobuild.SRC_HEADER_EXT:
            # arduino ide ignores files with .asm or .ASM extensions
            if ext.lower() == "asm":
                continue
            src_filter.append("+<*.%s>" % ext)
            if is_utility:
                src_filter.append("+<utility%s*.%s>" % (os.path.sep, ext))
```
https://github.com/platformio/platformio-core/blob/897844ebc172bd8a2e313bdb9011fc7d986863c2/platformio/builder/tools/piolib.py#L561C3-L561C3
```python
SRC_HEADER_EXT = ["h", "hpp"]
SRC_ASM_EXT = ["S", "spp", "SPP", "sx", "s", "asm", "ASM"]
SRC_C_EXT = ["c"]
SRC_CXX_EXT = ["cc", "cpp", "cxx", "c++"]
SRC_BUILD_EXT = SRC_C_EXT + SRC_CXX_EXT + SRC_ASM_EXT
```
https://github.com/platformio/platformio-core/blob/897844ebc172bd8a2e313bdb9011fc7d986863c2/platformio/builder/tools/piobuild.py#L31C1-L31C1

### 2. pioino.py - ino to cpp conversion
The .ino to .cpp conversion in `pioino.py` first searches for all .ino files (`FindInoNodes`) before converting it (`ConvertInoToCpp`).
The directory in which .ino-files will be searched is read from the "PROJECT_SRC_DIR" environment variable. `os.path.join(src_dir, "*.ino")` should be something like `os.path.join(src_dir, "**/*.ino")`. But i am not if this will cause any other problems.
```python
def FindInoNodes(env):
    src_dir = glob.escape(env.subst("$PROJECT_SRC_DIR"))
    return env.Glob(os.path.join(src_dir, "*.ino")) + env.Glob(
        os.path.join(src_dir, "*.pde")
    )
```
https://github.com/platformio/platformio-core/blob/897844ebc172bd8a2e313bdb9011fc7d986863c2/platformio/builder/tools/pioino.py#L225C9-L225C9

Would be great if anyone who is familiar with pio could fix this! Converting all files manually is an option, but not the best, as the files can then no longer be used with the Arduino IDE.

bimac, 2023-08-31:
Thanks for looking into this @PBahner! I tried switching the glob in `pioino.py` to recursive as per your suggestion. There seems to be some processing of the .ino files - but in the end it still fails:

``` gcc
> pio run -e v1
Processing v1 (platform: atmelsam; board: sparkfun_samd21_mini_usb; framework: arduino)
---------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelsam/sparkfun_samd21_mini_usb.html
PLATFORM: Atmel SAM (8.2.0) > SparkFun SAMD21 Mini Breakout
HARDWARE: SAMD21G18A 48MHz, 32KB RAM, 256KB Flash
DEBUG: Current (atmel-ice) External (atmel-ice, blackmagic, jlink)
PACKAGES: 
 - framework-arduino-samd-sparkfun @ 1.8.3 
 - framework-cmsis @ 1.40500.0 (4.5.0) 
 - framework-cmsis-atmel @ 1.2.2 
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
Converting v3.ino
/somewhere/project/src/v2/v2.ino:23:0: warning: "VERSION" redefined
 #define VERSION 2
 
/somewhere/project/src/v3/v3.ino:24:0: note: this is the location of the previous definition
 #define VERSION 3
 
/somewhere/project/src/v1/v1.ino:22:0: warning: "VERSION" redefined
 #define VERSION 1
 
/somewhere/project/src/v2/v2.ino:23:0: note: this is the location of the previous definition
 #define VERSION 2
 
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 9 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Linking .pio/build/v1/firmware.elf
.pio/build/v1/libFrameworkArduino.a(main.cpp.o): In function `main':
main.cpp:(.text.startup.main+0x22): undefined reference to `setup'
main.cpp:(.text.startup.main+0x26): undefined reference to `loop'
collect2: error: ld returned 1 exit status
*** [.pio/build/v1/firmware.elf] Error 1
================================= [FAILED] Took 0.55 seconds ================================
```

ivankravets, 2023-08-31:
I want to be transparent. We don't plan to invest any more into the INO format. This is a fake CPP format that adds more problems than benefits. What is more, the most popular IDEs do not support INO format. As a result, there are autocompletion and lining support. 

See a guide on migrating from INO to genuine CPP avoiding Arduino's crutchs. https://docs.platformio.org/en/latest/faq/ino-to-cpp.html

closed 2023-08-31 by ivankravets

bimac, 2023-08-31:
I agree about INO files being an issue - and I'm totally with you about the benefits of linting, autocompletion and everything. I don't use INO files for my own projects either and I wouldn't touch the Arduino IDE with a stick.

But there are still tons of projects out there that need/want to keep compatibility with the Arduino IDE. And I get it: There are plenty of not so technically-minded people that are overwhelmed by the complexity of a proper IDE like VSCode. These projects would greatly benefit from having a `platformio.ini` whilst keeping compatibility with Arduino - this way they could cater to both crowds.

Jason2866, 2023-11-26:
Maybe this https://github.com/platformio/platformio-core/pull/4297#issuecomment-1143469356
is a way?

ivankravets, 2023-12-01:
@Jason2866, thanks for the hint. This the solution https://github.com/platformio/platformio-core/pull/4297#issuecomment-1143469356