platformio/platformio-core #4739: Upload and Monitor in CLion fails with termios.error: (25, 'Inappropriate ioctl for device')

mizraith, 2023-09-07:
What kind of issue is this?
Monitor does not work.  CLion 2023.2.1.   Platformio upgrades.

- [x ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

Using Upload and Monitor on a Mac M1.    The upload works (serial port doesn't seem busy).  Seen this in other forums as well.

error: Traceback (most recent call last):
  File "/opt/homebrew/lib/python3.11/site-packages/platformio/__main__.py", line 103, in main
    cli()  # pylint: disable=no-value-for-parameter
    ^^^^^
  File "/opt/homebrew/lib/python3.11/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.11/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.11/site-packages/platformio/cli.py", line 85, in invoke
    return super().invoke(ctx)
           ^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.11/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.11/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.11/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.11/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.11/site-packages/platformio/run/cli.py", line 145, in cli
    process_env(
  File "/opt/homebrew/lib/python3.11/site-packages/platformio/run/cli.py", line 211, in process_env
    ctx.invoke(
  File "/opt/homebrew/lib/python3.11/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.11/site-packages/platformio/device/monitor/command.py", line 152, in device_monitor_cmd
    start_terminal(options)
  File "/opt/homebrew/lib/python3.11/site-packages/platformio/device/monitor/terminal.py", line 54, in start_terminal
    term = new_terminal(options)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.11/site-packages/platformio/device/monitor/terminal.py", line 102, in new_terminal
    term = Terminal(
           ^^^^^^^^^
  File "/opt/homebrew/lib/python3.11/site-packages/platformio/device/monitor/terminal.py", line 28, in __init__
    super().__init__(*args, **kwargs)
  File "/opt/homebrew/lib/python3.11/site-packages/serial/tools/miniterm.py", line 395, in __init__
    self.console = Console()
                   ^^^^^^^^^
  File "/opt/homebrew/lib/python3.11/site-packages/serial/tools/miniterm.py", line 198, in __init__
    self.old = termios.tcgetattr(self.fd)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^
termios.error: (25, 'Inappropriate ioctl for device')

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `python -m pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`):

### Description of problem
Monitor fails after upload.

#### Steps to Reproduce

1.  Upload arduino project with monitor enabled
2.  Observe the stack trace
3. 3.  Verified serial port baud rate, port settings, etc.

### Actual Results
See above stack trace

LandryNorris, 2023-09-08:
This is [an issue in CLion](https://youtrack.jetbrains.com/issue/CPP-19591/CLion-Tools-PlatformIO-Monitor-Error-25-Inappropriate-ioctl-for-device). Running "pio device monitor" via 'Run Anything' avoids a crash, but serial output is not rendered due to [this bug](https://youtrack.jetbrains.com/issue/CPP-33401/Serial-Port-Monitor-utf-8-support). Support for PlatformIO in CLion is still a bit of a work in progress. 

A workaround that I use is to run 'pio device monitor' in the CLion terminal, which works well.

ivankravets, 2023-09-22:
Duplicate of https://youtrack.jetbrains.com/issue/CPP-19591/CLion-Tools-PlatformIO-Monitor-Error-25-Inappropriate-ioctl-for-device

closed 2023-09-22 by ivankravets