platformio/platformio-core #4741: toolchain-gccarmnoneeabi  version 1.120201.221222  triggers Blocked insecure item errors

ellensp, 2023-09-07:
### Configuration

Confirmed on Ubuntu and Mac

PlatformIO Core, version 6.1.11

### Description of problem

Latest https://registry.platformio.org/tools/platformio/toolchain-gccarmnoneeabi  version 1.120201.221222 https://registry.platformio.org/tools/platformio/toolchain-gccarmnoneeabi/versions will not install
correctly

[image]

Installing this triggers "Blocked insecure item `%s` from TAR archive" % item.name, in

https://github.com/platformio/platformio-core/blob/develop/platformio/package/unpack.py#L84-L98

The archive seems to be damaged in some way.

### Actual Results

[image]

### Expected Results

No Blocked insecure item errors.

### If problems with PlatformIO Build System:

### Additional info

This was notced by Marlin firmware user, code and platoform ini files are huge, and not really relevant to this issue.

maxgerhardt, 2023-09-18:
FYI this was added in https://github.com/platformio/platformio-core/commit/2388b2a62be5c220736dbf205bb33d413fa04cea almost 4 years ago. I think the toolchains come from the GNU ARM download directly.

ivankravets, 2023-09-22:
We have just removed the broken `1.120201.221222` version. Please try the recently published [1.120301.0](https://registry.platformio.org/tools/platformio/toolchain-gccarmnoneeabi)

closed 2023-09-22 by ivankravets

ellensp, 2023-09-22:
Issue is still present with 1.120301.0

maxgerhardt, 2023-09-22:
No, it does work with the new package on my machine.
```
$ pio pkg install -g -t "toolchain-gccarmnoneeabi@1.120301.0"
Tool Manager: Installing toolchain-gccarmnoneeabi @ 1.120301.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: toolchain-gccarmnoneeabi@1.120301.0 has been installed!
```

```
$ pio system info
--------------------------  --------------------------------------------
PlatformIO Core             6.1.11
Python                      3.11.4-final.0
System Type                 linux_x86_64
Platform                    Linux-6.2.0-32-generic-x86_64-with-glibc2.37
```

ellensp, 2023-09-22:
secondary linking warnings that I thought stemmed from this issue are still present.  These threw me

This issue is resolved.

maxgerhardt, 2023-09-22:
Linking warnings? In Arduino-Pico we also noticed we need `build_flags = -Wl,--no-warn-rwx-segments` to supress a supurious warning.

ellensp, 2023-09-22:
well getting off topic here but...

Getting these linker warnings 

warning: _close is not implemented and will always fail
warning: _fstat is not implemented and will always fail
warning: _getpid is not implemented and will always fail
warning: _isatty is not implemented and will always fail
warning: _kill is not implemented and will always fail
warning: _lseek is not implemented and will always fail
warning: _read is not implemented and will always fail
warning: _write is not implemented and will always fail

If I revert toolchain-gccarmnoneeabi to the older release
pio pkg uninstall -g -t "toolchain-gccarmnoneeabi@1.120301.0"
pio pkg install -g -t "toolchain-gccarmnoneeabi@1.100301.220327"

The warnings are not present

ivankravets, 2023-09-22:
@ellensp , what is your development `platform`?