platformio/platformio-core #4750: PlatformIO serial montior reporting wrong version in Clion IDE

Bgreening, 2023-09-24:
What kind of issue is this?

When launching PlatformIO serial montior from Clion IDE I am prompted to '* Verify that you have the latest version of PlatformIO using - - - `python -m pip install -U platformio` command. See ### Actual Results

**PlatformIO Version** (`platformio --version`):
PlatformIO Core, version 6.1.11

### Description of problem
PlatformIO serial montior reporting wrong version in Clion IDE
Am I missing something when a verify the PlatformIO version? Any advice would be helpful.
Bgreening

#### Steps to Reproduce

1. Launch PlatformIO serial monitor from Clion IDE

### Actual Results
Ubuntu 22.04 and Clion 2023 2.2 and launching PlatformIO serial montior from Clion IDE I am prompted to * Verify that you have the latest version of PlatformIO using `python -m pip install -U platformio` command. 

From the terminal window I run python3 -m pip install -U platformio. I get the the following:

Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: platformio in ./.local/lib/python3.10/site-packages (6.1.11)
Requirement already satisfied: semantic-version==2.10.* in ./.local/lib/python3.10/site-packages (from platformio) (2.10.0)
Requirement already satisfied: uvicorn<0.24,>=0.16 in ./.local/lib/python3.10/site-packages (from platformio) (0.23.2)
Requirement already satisfied: pyelftools==0.29 in ./.local/lib/python3.10/site-packages (from platformio) (0.29)
Requirement already satisfied: bottle==0.12.* in ./.local/lib/python3.10/site-packages (from platformio) (0.12.25)
Requirement already satisfied: marshmallow==3.* in ./.local/lib/python3.10/site-packages (from platformio) (3.20.1)
Requirement already satisfied: colorama in /usr/lib/python3/dist-packages (from platformio) (0.4.4)
Requirement already satisfied: click<=8.2,>=8.0.4 in ./.local/lib/python3.10/site-packages (from platformio) (8.1.7)
Requirement already satisfied: pyserial==3.5.* in ./.local/lib/python3.10/site-packages (from platformio) (3.5)
Requirement already satisfied: wsproto==1.* in ./.local/lib/python3.10/site-packages (from platformio) (1.2.0)
Requirement already satisfied: requests==2.* in /usr/lib/python3/dist-packages (from platformio) (2.25.1)
Requirement already satisfied: starlette<0.32,>=0.19 in ./.local/lib/python3.10/site-packages (from platformio) (0.31.1)
Requirement already satisfied: tabulate==0.* in ./.local/lib/python3.10/site-packages (from platformio) (0.9.0)
Requirement already satisfied: ajsonrpc==1.2.* in ./.local/lib/python3.10/site-packages (from platformio) (1.2.0)
Requirement already satisfied: packaging>=17.0 in ./.local/lib/python3.10/site-packages (from marshmallow==3.*->platformio) (23.1)
Requirement already satisfied: h11<1,>=0.9.0 in ./.local/lib/python3.10/site-packages (from wsproto==1.*->platformio) (0.14.0)
Requirement already satisfied: anyio<5,>=3.4.0 in ./.local/lib/python3.10/site-packages (from starlette<0.32,>=0.19->platformio) (4.0.0)
Requirement already satisfied: typing-extensions>=4.0 in ./.local/lib/python3.10/site-packages (from uvicorn<0.24,>=0.16->platformio) (4.7.1)
Requirement already satisfied: sniffio>=1.1 in ./.local/lib/python3.10/site-packages (from anyio<5,>=3.4.0->starlette<0.32,>=0.19->platformio) (1.3.0)
Requirement already satisfied: exceptiongroup>=1.0.2 in ./.local/lib/python3.10/site-packages (from anyio<5,>=3.4.0->starlette<0.32,>=0.19->platformio) (1.1.3)
Requirement already satisfied: idna>=2.8 in /usr/lib/python3/dist-packages (from anyio<5,>=3.4.0->starlette<0.32,>=0.19->platformio) (3.3)

It looks like I have the current PlatformIO version.

### Expected Results
Serial monitor should start in Clion run window.

**The content of `platformio.ini`:**
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:nodemcu-32s]
platform = espressif32
board = nodemcu-32s
framework = arduino
monitor_speed = 115200
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>
#include <WiFi.h>

/*
  Example from WiFi > WiFiScan
  Complete details at https://RandomNerdTutorials.com/esp32-useful-wi-fi-functions-arduino/
*/

void setup() {
    Serial.begin(115200);

    // Set WiFi to station mode and disconnect from an AP if it was previously connected
    WiFi.mode(WIFI_STA);
    WiFi.disconnect();
    delay(100);

    Serial.println("Setup done");}

void loop() {
    Serial.println("scan start");

    // WiFi.scanNetworks will return the number of networks found
    int n = WiFi.scanNetworks();
    Serial.println("scan done");
    if (n == 0) {
        Serial.println("no networks found");
    } else {
        Serial.print(n);
        Serial.println(" networks found");
        for (int i = 0; i < n; ++i) {
            // Print SSID and RSSI for each network found
            Serial.print(i + 1);
            Serial.print(": ");
            Serial.print(WiFi.SSID(i));
            Serial.print(" (");
            Serial.print(WiFi.RSSI(i));
            Serial.print(")");
            Serial.println((WiFi.encryptionType(i) == WIFI_AUTH_OPEN)?" ":"*");
            delay(10);
        }
    }
    Serial.println("");

    // Wait a bit before scanning again
    delay(5000);}
```

### Additional info

ivankravets, 2023-11-29:
cc @elmot

elmot, 2023-12-27:
I don't believe that this issue is somehow related to CLion

ivankravets, 2023-12-30:
@Bgreening , could you provide a screenshot of this issue?