platformio/platformio-core #4751: Cannot create new PlatformIO project in CLion

drewmutt, 2023-09-25:
### Configuration

**Operating system**:
Mac (M1) 12.5.1

**PlatformIO Version** (`platformio --version`):
6.1.11

**Plugin Version**:
233.6745.300 (had same issue in prev versions)

**CLion Version**:
CLion 2023.3 EAP (same issue in 2023.2.2)

### Description of problem
The first step of the create new PlatformIO project, fails. Preventing me from creating new projects in CLion.

#### Steps to Reproduce
1. File > New Project > PlatformIO

### Actual Results
Behold the text "Gathering info..." in the board list, and the following error below it:
`Use JsonReaderEx.setLenient(true) to accept malformed JSON at line 1 column 2`

The "Create" button is expectingly gray, and progress cannot continue.

### Expected Results
Display the list of boards, and allow the user to select one to continue.

### Additional info
Looking into the [source](https://github.com/JetBrains/intellij-plugins/blob/master/platformio/src/com/jetbrains/cidr/cpp/embedded/platformio/project/PlatformioProjectSettingsStep.kt), I found a call to `pio boards --json-output`.

Executing it appears to work correctly, and the JSON doesn't appear to be corrupt.

ak1987, 2023-10-22:
If your any CLI input (like `pio --version`) comes with warning like this:
`.platformio/penv/lib/python3.9/site-packages/urllib3/__init__.py:34: NotOpenSSLWarning: urllib3 v2.0 only supports OpenSSL 1.1.1+, currently the 'ssl' module is compiled with 'LibreSSL 2.8.3'. See: https://github.com/urllib3/urllib3/issues/3020`
You should try downgrade library with following command:
```
. ~/.platformio/penv/bin/activate
pip install urllib3==1.26.15
```
JFYI: [https://github.com/platformio/platformio-core/issues/4614](url)
_TLDR: Warning comes in board list output causing parser fail_

ivankravets, 2023-11-29:
@ak1987 , thanks for the solution. It the right solution to check `pio boards` command.

closed 2023-11-29 by ivankravets

Quantriple, 2024-03-16:
when i run "pio --version", show  "PlatformIO Core, version 6.1.11", but  clion has been staying gathering info.......,could you please help me ......