platformio/platformio-core #4757: Allow multiple linker scripts in `ldscript` for a board.

Ariakenom, 2023-10-09:
I'm using multiple `-Wl,-T,mem.ld` in `build_flags` but according to this message I should rather be using `ldscript`. Except it doesn't allow for more than one script.
https://github.com/platformio/platform-ststm32/blob/f05fdfbd948a705e99f241b8e3be35bf64e019b9/builder/main.py#L355

Previously asked and answered on the forum.
https://community.platformio.org/t/multiple-scripts-in-board-build-ldscript/36062
https://github.com/platformio/platformio-core/blob/7229e1cce4a06cc2f3b2cb6fbcb2ddddb77b0320/platformio/builder/tools/piobuild.py#L64-L66

Ariakenom, 2023-11-29:
> same here! But I think this ticket should be in platform-ststm32

The issue is about this line in platformio-core: https://github.com/platformio/platformio-core/blob/b238c55e53a910c158f4c4766b24f417e69de535/platformio/builder/tools/piobuild.py#L66

ivankravets, 2023-11-29:
A temporary solution is to use Extra Scripting. See example https://docs.platformio.org/en/latest/scripting/examples/extra_linker_flags.html

Ariakenom, 2023-11-29:
No workaround needed. Multiple `-Wl,-T,mem.ld` works fine, it's just that platform-ststm32 is complaining. Maybe @fbenamrouche meant platform-ststm32 should stop complaining.