platformio/platformio-core #4764: Install on ubuntu: Failed to open the app

mshenouda, 2023-10-20:
Traceback (most recent call last):
  File "/usr/bin/platformio", line 33, in <module>
    sys.exit(load_entry_point('platformio==4.3.4', 'console_scripts', 'platformio')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/bin/platformio", line 25, in importlib_load_entry_point
    return next(matches).load()
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/metadata/__init__.py", line 202, in load
    module = import_module(match.group('module'))
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/usr/lib/python3/dist-packages/platformio/__main__.py", line 66, in <module>
    @cli.resultcallback()
     ^^^^^^^^^^^^^^^^^^
AttributeError: 'PlatformioCLI' object has no attribute 'resultcallback'. Did you mean: 'result_callback'?

SakuraPuare, 2023-11-25:
do not use `sudo apt install platformio` to install platformio

using 
```bash
curl -fsSL -o get-platformio.py https://raw.githubusercontent.com/platformio/platformio-core-installer/master/get-platformio.py
python3 get-platformio.py
```

according https://docs.platformio.org/en/latest/core/installation/methods/installer-script.html

ivankravets, 2023-11-25:
@SakuraPuare , thanks for the solution!

closed 2023-11-25 by ivankravets

NathanLibutan, 2024-09-07:
> do not use `sudo apt install platformio` to install platformio
> 
> using
> 
> ```shell
> curl -fsSL -o get-platformio.py https://raw.githubusercontent.com/platformio/platformio-core-installer/master/get-platformio.py
> python3 get-platformio.py
> ```
> 
> according https://docs.platformio.org/en/latest/core/installation/methods/installer-script.html

I'm running into this same problem but I'm using curl... Not sure if anyone else has had the same issue?

ivankravets, 2024-09-09:
See https://github.com/platformio/platformio-core-installer/issues/85

shekhov, 2025-12-30:
> > platformio==4.3.4
> 
> It seems you have multiple PIO Core in the system. 4.3.4 was released almost 10 years ago. 6.1 is the latest stable version. Please uninstall legacy conflicted versions

Oh my.. thanks a lot for catching this :)

jl2035, 2026-09-15:
I fixed this by not using the script:

```
virutalenv venv
source venv/bin/activate
pip install platformio
```