platformio/platformio-core #4781: Allow configuring registry host url

bsdayo, 2023-11-09:
What kind of issue is this?

- [x] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 11

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 6.1.11

### Description of problem

The current PlatformIO registry host url ([registry.platformio.org](https://registry.platformio.org)) seems to be hardcoded in the source code and cannot be configured. It would be handy if it could be configured.

As a user in China, I would like to thank the PlatformIO team for their efforts in trying to provide a stable service for Chinese users. However, in many areas of China, the registry is still very slow to access. Personally, I'm not optimistic about building a CDN node specifically for China (mentioned in #4345), because China's policy is very restrictive on foreign services, and as long as the server is outside of China, it may be blocked by GFW at any time.

So if you can configure the registry host url, you can build your own server to reverse proxy the registry, and then point the registry host url to your own server. This increases flexibility because even if PlatformIO's official CDN node is blocked or hard to access, users can continue to access the Registry using their own reverse proxy server.

### Additional info

I have recommended PlatformIO more than once to my close friends, but the biggest resistance is the network problem. It would be great if this problem could be solved completely.

villivateur, 2023-11-17:
I have the same request.

karlzheng, 2024-03-10:
I have the same request.