platformio/platformio-core #4796: Member function override error with PIO build

divStar, 2023-11-27:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: MacOS 14.0 with XCode 15.0.1

**PlatformIO Version** (`platformio --version`): `PlatformIO Core, version 6.1.11`

### Description of problem
I cannot override a pure virtual member function as I think I should be able to. While the CLion editor does not show any errors, PlatformIO just refuses to compile the code.

My `ILed.h` (abstract class / interface) contains this pure virtual member function:
```cpp
[[nodiscard]] virtual const uint8_t (&getColor())[4] = 0;
```

My `HardwareLed.h` (concrete implementation of `ILed`) contains this supposed override to the pure virtual member function above:
```cpp
[[nodiscard]] const uint8_t (&getColor())[4] override;
```

However: when I run `pio run` from the git repository root, it fails.
```shell
...
include/HardwareLed.h:12:46: error: expected ';' at end of member declaration
   [[nodiscard]] const uint8_t (&getColor())[4] override;
                                              ^
include/HardwareLed.h:12:48: error: 'override' does not name a type; did you mean 'ctermid'?
   [[nodiscard]] const uint8_t (&getColor())[4] override;
                                                ^~~~~~~~
                                                ctermid
...
```

#### Steps to Reproduce
I have made a dedicated minimal reproducible: https://github.com/divStar/minimal-example-getter-problem .

### Actual Results
```shell
pio run
Processing zeroUSB (platform: atmelsam; board: zeroUSB; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelsam/zeroUSB.html
PLATFORM: Atmel SAM (8.2.0) > Arduino Zero (USB Native Port)
HARDWARE: SAMD21G18A 48MHz, 32KB RAM, 256KB Flash
DEBUG: Current (atmel-ice) External (atmel-ice, blackmagic, jlink)
PACKAGES: 
 - framework-arduino-samd @ 1.8.13 
 - framework-cmsis @ 1.40500.0 (4.5.0) 
 - framework-cmsis-atmel @ 1.2.2 
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 12 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/zeroUSB/src/HardwareLed.cpp.o
In file included from src/HardwareLed.cpp:1:0:
include/HardwareLed.h:12:46: error: expected ';' at end of member declaration
   [[nodiscard]] const uint8_t (&getColor())[4] override;
                                              ^
include/HardwareLed.h:12:48: error: 'override' does not name a type; did you mean 'ctermid'?
   [[nodiscard]] const uint8_t (&getColor())[4] override;
                                                ^~~~~~~~
                                                ctermid
*** [.pio/build/zeroUSB/src/HardwareLed.cpp.o] Error 1
====================================================================================== [FAILED] Took 0.52 seconds ====================================================================================== 
```

### Expected Results
There should be no error when compiling.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
See: https://github.com/divStar/minimal-example-getter-problem/blob/master/platformio.ini
```ini
[platformio]
description = Minimal example with failing reference getter
default_envs = zeroUSB

[env:zeroUSB]
platform = atmelsam
board = zeroUSB
framework = arduino
build_flags = -std=gnu++17
build_unflags = -std=gnu++11
```

**Source file to reproduce issue:**
Git repository: https://github.com/divStar/minimal-example-getter-problem/tree/master .

### Additional info
Removing the `override` from `[[nodiscard]] const uint8_t (&getColor())[4] override;` line in [`HardwareLed.h`](https://github.com/divStar/minimal-example-getter-problem/blob/d96d87b896b3dedb77b069012bad9288d3c5b2b5/include/HardwareLed.h#L12) makes PlatformIO compile successfully.

Also my original files contain multiple namespaces, but it's not a namespace issue it seems.

Even removing the following lines from the `platformio.ini` file does not have an impact on the errors:
```ini
build_flags = -std=gnu++17
build_unflags = -std=gnu++11
```

ivankravets, 2023-11-27:
Thanks, @valeros, for the explanation.

closed 2023-11-27 by ivankravets

divStar, 2023-11-29:
@valeros thank you! It seems it only is a bug with GCC, but not if compiled using clang or icx. Sadly I am not that well-versed in C++ in general, so I am learning, which is why I decided to [ask on Stackoverflow](https://stackoverflow.com/questions/77567844/overriding-a-method-returning-a-reference-to-an-array-does-not-work-in-gcc-but) and hopefully someone can tell me how it's done in gcc or if it shouldn't be done at all (I enhanced your example a bit - `getColor` without the `2` could be an alternative, though it does not limit the array to just 4 items).