platformio/platformio-core #4810: Custom launch configuration gets overwritten in launch.json

AnthonyDTU, 2023-12-20:
Issue created based on my comment from issue #3824 

My goal is to create a "***PIO Attach***" launch configuration. However, when the platformio.ini file is updated/saved, the launch.json files is overwritten. I believe that this is handled by this file: https://github.com/platformio/platformio-core/blob/develop/platformio/project/integration/tpls/vscode/.vscode/launch.json.tpl.

In that file, the code which is responsible for "saving" external configurations, looks for all configurations ***not*** of the type: "platformio-debug" (line 65). However, my ***PIO Attach*** configuration is of that type, and it therefore does not get saved, resulting in it getting overwritten. 

Is there a possibility to only look for the three launch configurations: ***PIO Debug***, ***PIO Debug (skip Pre-Debug)*** and ***PIO Debug (without uploading)***?

ivankravets, 2023-12-31:
Please re-test with `pio upgrade --dev`.

closed 2023-12-31 by ivankravets

AnthonyDTU, 2024-01-02:
It works as desired, thank you

commit b181406a (2023-12-31): Resolve an issue where custom debug configurations were being inadvertently overwritten in VSCode's ``launch.json`` // Resolve #4810