platformio/platformio-core #4817: Having a custom board definition in boards folder breaks debugging of native unit tests

Strooom, 2024-01-02:
What kind of issue is this?  **PlatformIO Core**.

### Configuration

**Operating system**: Win11 x64
**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 6.1.11

### Description of problem
Having a folder 'boards' with a custom board definition, results in failing the debugger to start debugging a native unit test.
Removing the folder -> unit tests can be debugged
Restoring the folder -> unit tests cannot be debugged

#### Steps to Reproduce 1/3

1. download repo https://github.com/Strooom/NewProject3PIO
2. select (the only) environment [env:some_generic_unittests]
3. start debugging

### Actual Results
nothing happens

### Expected Results
debugger should start

#### Steps to Reproduce 2/3
remove boards folder -> debugger will start

#### Steps to Reproduce 3/3
restore boards folder -> problem returns

### Additional info
I think this is related to forum topic https://community.platformio.org/t/attributeerror-dict-object-has-no-attribute-get-debug-tool-name/26335

ivankravets, 2024-01-04:
Could you try the latest development version? `$ pio upgrade --dev`

We will release 6.1.12 in the next days.

Strooom, 2024-01-05:
In PlatformIO Core, version 6.1.12b1 it works correctly. 
Thank you!

closed 2024-01-05 by Strooom