platformio/platformio-core #4820: Code Inspect Error

rvxfahim, 2024-01-06:
Why is the following line of code labelled as "CWE-788:Array 'ip[4]' accessed at index 4, which is out of bounds."
ip[4] = {0};

valeros, 2024-01-10:
Hi @rvxfahim, your code snippet is not enough to evaluate the correctness of the reported defect. Please attach here a minimal PlatformIO project to reproduce the issue.

rvxfahim, 2024-01-10:
> Hi @rvxfahim, your code snippet is not enough to evaluate the correctness of the reported defect. Please attach here a minimal PlatformIO project to reproduce the issue.

uint8_t ip[4] = {192, 168, 31, 4};
void BLEipCallBack::onWrite(BLECharacteristic *pCharacteristic)
{
    std::string value = pCharacteristic->getValue();
    // parse the string to get the IP address
    // the string is in the format "xxx.xxx.xxx.xxx"
    // where x is a number
    // the IP address is stored in the global variable ip
    ip[4] = {0};
    int segmentIndex = 0;
    std::string segment = "";
    for (char c : value)
    {
        if (c == '.')
        {
            ip[segmentIndex] = std::stoi(segment);
            segmentIndex++;
            segment = "";
        }
        else
        {
            segment += c;
        }
    }
    // handle the last segment
    ip[segmentIndex] = std::stoi(segment);
    // store the blob 
    NVS.setBlob("ip", ip, sizeof(ip));
    // set the IP address
    server[0] = ip[0];
    server[1] = ip[1];
    server[2] = ip[2];
    server[3] = ip[3];
    nh.getHardware()->setConnection(server, serverPort);
    Serial.println("IP address changed to:");
    Serial.print(ip[0]);
    Serial.print(".");
    Serial.print(ip[1]);
    Serial.print(".");
    Serial.print(ip[2]);
    Serial.print(".");
    Serial.println(ip[3]);
}

valeros, 2024-01-11:
You've declared an array of size 4, the array indexing in C/C++ starts from 0, so the lowest index is 0 and the highest index is 3. Executing `ip[4]` is an "index out of bounds error" which is correctly reported by Cppcheck.

closed 2024-01-11 by valeros