platformio/platformio-core #4825: [Feature request]: debug support for vim?

ubaldot, 2024-01-07:
I just discovered that it is possible to debug with platoformio and I love such a 1-click concept! :smiley:
However, I have seen that it is possible with VSCode and I am wondering if it is also possible with Vim. AFAIK, there is [vimspector](https://github.com/puremourning/vimspector) which seems to act as a Debug Adapter Protocol (DAP) client, or the builtin `:termdebug.`

ivankravets, 2024-01-09:
Debugging should work. I previously have used Vim/Emacs for so many years and tested PlatformIO Debugging. The CLI command is `pio debug --interface gdb -x .pioinit`

ubaldot, 2024-01-10:
Ok, would that be possible to add a guideline in the docs? I could try in the meantime 🙂

ivankravets, 2024-01-10:
Yes, we would be thankful for the PR. The raw page is hosted here https://github.com/platformio/platformio-docs/blob/develop/integration/ide/vim.rst