platformio/platformio-core #4827: Release 6.1.12 breaks installation through github actions

Strooom, 2024-01-10:
What kind of issue is this?  **PlatformIO Core**.

I am running a CI/CD workflow on github actions, which installs (the most recent version of) PlatformIO.
After the upgrade to 6.1.12, this installation fails.
On my development machine (Win64 v11) the installation/upgrade worked ok.

repository : https://github.com/Strooom/MuMo-v2-Node-SW
workflow run #22, still using 6.1.11 ran OK
workflow run #23, installing 6.1.12 failed the installation :  looks like it fails on/after the installation of the ststm32 package

Here is the log of the installation that failed
```
Run pip install --upgrade platformio
  pip install --upgrade platformio
  pio pkg install --environment dummy
  shell: /usr/bin/bash -e {0}
  env:
    pythonLocation: /opt/hostedtoolcache/Python/3.9.18/x64
    PKG_CONFIG_PATH: /opt/hostedtoolcache/Python/3.9.18/x64/lib/pkgconfig
    Python_ROOT_DIR: /opt/hostedtoolcache/Python/3.9.18/x64
    Python[2](https://github.com/Strooom/MuMo-v2-Node-SW/actions/runs/7479180355/job/20355781257#step:10:2)_ROOT_DIR: /opt/hostedtoolcache/Python/[3](https://github.com/Strooom/MuMo-v2-Node-SW/actions/runs/7479180355/job/20355781257#step:10:3).9.18/x6[4](https://github.com/Strooom/MuMo-v2-Node-SW/actions/runs/7479180355/job/20355781257#step:10:4)
    Python3_ROOT_DIR: /opt/hostedtoolcache/Python/3.9.18/x64
    LD_LIBRARY_PATH: /opt/hostedtoolcache/Python/3.9.18/x64/lib
Collecting platformio
  Downloading platformio-6.1.12.tar.gz (237 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 237.8/237.8 kB 8.[5](https://github.com/Strooom/MuMo-v2-Node-SW/actions/runs/7479180355/job/20355781257#step:10:5) MB/s eta 0:00:00
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Collecting bottle==0.12.* (from platformio)
  Using cached bottle-0.12.25-py3-none-any.whl (90 kB)
Collecting click<9,>=8.0.4 (from platformio)
  Using cached click-8.1.7-py3-none-any.whl.metadata (3.0 kB)
Collecting colorama (from platformio)
  Using cached colorama-0.4.[6](https://github.com/Strooom/MuMo-v2-Node-SW/actions/runs/7479180355/job/20355781257#step:10:6)-py2.py3-none-any.whl (25 kB)
Collecting marshmallow==3.* (from platformio)
  Downloading marshmallow-3.20.2-py3-none-any.whl.metadata ([7](https://github.com/Strooom/MuMo-v2-Node-SW/actions/runs/7479180355/job/20355781257#step:10:7).5 kB)
Collecting pyelftools==0.30 (from platformio)
  Downloading pyelftools-0.30-py2.py3-none-any.whl.metadata (3[8](https://github.com/Strooom/MuMo-v2-Node-SW/actions/runs/7479180355/job/20355781257#step:10:8)1 bytes)
Collecting pyserial==3.5.* (from platformio)
  Using cached pyserial-3.5-py2.py3-none-any.whl ([9](https://github.com/Strooom/MuMo-v2-Node-SW/actions/runs/7479180355/job/20355781257#step:10:9)0 kB)
Collecting requests==2.* (from platformio)
  Using cached requests-2.31.0-py3-none-any.whl.metadata (4.6 kB)
Collecting semantic-version==2.[10](https://github.com/Strooom/MuMo-v2-Node-SW/actions/runs/7479180355/job/20355781257#step:10:10).* (from platformio)
  Using cached semantic_version-2.10.0-py2.py3-none-any.whl (15 kB)
Collecting tabulate==0.* (from platformio)
  Using cached tabulate-0.9.0-py3-none-any.whl (35 kB)
Collecting ajsonrpc==1.2.* (from platformio)
  Using cached ajsonrpc-1.2.0-py3-none-any.whl (22 kB)
Collecting starlette<0.35,>=0.19 (from platformio)
  Downloading starlette-0.34.0-py3-none-any.whl.metadata (5.8 kB)
Collecting uvicorn<0.26,>=0.16 (from platformio)
  Downloading uvicorn-0.25.0-py3-none-any.whl.metadata (6.4 kB)
Collecting wsproto==1.* (from platformio)
  Using cached wsproto-1.2.0-py3-none-any.whl (24 kB)
Collecting packaging>=17.0 (from marshmallow==3.*->platformio)
  Using cached packaging-23.2-py3-none-any.whl.metadata (3.2 kB)
Collecting charset-normalizer<4,>=2 (from requests==2.*->platformio)
  Using cached charset_normalizer-3.3.2-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (33 kB)
Collecting idna<4,>=2.5 (from requests==2.*->platformio)
  Using cached idna-3.6-py3-none-any.whl.metadata (9.9 kB)
Collecting urllib3<3,>=1.21.1 (from requests==2.*->platformio)
  Using cached urllib3-2.1.0-py3-none-any.whl.metadata (6.4 kB)
Collecting certifi>=2017.4.17 (from requests==2.*->platformio)
  Using cached certifi-2023.[11](https://github.com/Strooom/MuMo-v2-Node-SW/actions/runs/7479180355/job/20355781257#step:10:11).17-py3-none-any.whl.metadata (2.2 kB)
Collecting h11<1,>=0.9.0 (from wsproto==1.*->platformio)
  Using cached h11-0.14.0-py3-none-any.whl (58 kB)
[... 75 lines cut ...]
  File "/opt/hostedtoolcache/Python/3.9.18/x64/lib/python3.9/site-packages/platformio/platform/base.py", line 185, in configure_project_packages
    self.configure_default_packages(options, targets or [])
  File "/home/runner/.platformio/platforms/ststm32/platform.py", line 29, in configure_default_packages
    board_config = self.board_config(board)
  File "/opt/hostedtoolcache/Python/3.9.18/x64/lib/python3.9/site-packages/platformio/platform/base.py", line [17](https://github.com/Strooom/MuMo-v2-Node-SW/actions/runs/7479180355/job/20355781257#step:10:18)2, in board_config
    assert id_
AssertionError

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `python -m pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

Error: Process completed with exit code 1.
```

Strooom, 2024-01-10:
changed the command to install PlatformIO to `pip install platformio==6.1.11` and now the workflow is running again. But of course it would be better to be able to run the latest version.

ivankravets, 2024-01-11:
The issue is caused by a custom board. Do you have custom board JSON in `boards` folder?

ivankravets, 2024-01-11:
The `ststm32` dev-platform does not support running an environment without `board` definition. See https://github.com/platformio/platform-ststm32/blob/develop/platform.py#L29

It was previously a bug in PIO Core that didn't raise the exception.

Regarding your `dummy` configuration, just add any `board = ...`

closed 2024-01-11 by ivankravets

Strooom, 2024-01-11:
I can confirm that when adding the custom board in platformio.ini, the installation now works again. Thanks Ivan.

artyomsv, 2024-01-27:
I`m facing the same issue using version 6.1.13. In my scenario, I do not want to define -e and want to install packages for all the environments (20+), but I am getting same exception

```
  File "/opt/hostedtoolcache/Python/3.9.18/x64/lib/python3.9/site-packages/platformio/platform/base.py", line [17](https://github.com/Strooom/MuMo-v2-Node-SW/actions/runs/7479180355/job/20355781257#step:10:18)2, in board_config
    assert id_
AssertionError
```

ivankravets, 2024-01-27:
Could you share your `platformio.ini`?

artyomsv, 2024-01-27:
> Could you share your `platformio.ini`?

```ini
#
# Marlin Firmware
# PlatformIO Configuration File
#
# For detailed documentation with EXAMPLES:
#
# https://docs.platformio.org/en/latest/projectconf/index.html
#

# Automatic targets - enable auto-uploading
#targets = upload

[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = mega2560
include_dir  = Marlin
extra_configs =
    Marlin/config.ini
    ini/avr.ini
    ini/due.ini
    ini/esp32.ini
    ini/features.ini
    ini/lpc176x.ini
    ini/native.ini
    ini/samd21.ini
    ini/samd51.ini
    ini/stm32-common.ini
    ini/stm32f0.ini
    ini/stm32f1-maple.ini
    ini/stm32f1.ini
    ini/stm32f4.ini
    ini/stm32f7.ini
    ini/stm32h7.ini
    ini/stm32g0.ini
    ini/teensy.ini
    ini/renamed.ini

#
# The 'common' section applies to most Marlin builds.
#
# By default platformio build will abort after 5 errors.
# Remove '-fmax-errors=5' from build_flags below to see all.
#
[common]
build_flags        = -g3 -D__MARLIN_FIRMWARE__ -DNDEBUG
                     -fmax-errors=5
extra_scripts      =
  pre:buildroot/share/PlatformIO/scripts/configuration.py
  pre:buildroot/share/PlatformIO/scripts/common-dependencies.py
  pre:buildroot/share/PlatformIO/scripts/common-cxxflags.py
  pre:buildroot/share/PlatformIO/scripts/preflight-checks.py
  post:buildroot/share/PlatformIO/scripts/common-dependencies-post.py
lib_deps           =
default_src_filter = +<src/*> -<src/config> -<src/HAL> +<src/HAL/shared> -<src/tests>
  -<src/lcd/HD44780> -<src/lcd/TFTGLCD> -<src/lcd/dogm> -<src/lcd/tft> -<src/lcd/tft_io>
  -<src/HAL/STM32/tft> -<src/HAL/STM32F1/tft>
  -<src/lcd/e3v2/common> -<src/lcd/e3v2/creality> -<src/lcd/e3v2/proui> -<src/lcd/e3v2/jyersui> -<src/lcd/e3v2/marlinui>
  -<src/lcd/menu>
  -<src/lcd/menu/game/game.cpp> -<src/lcd/menu/game/brickout.cpp> -<src/lcd/menu/game/invaders.cpp>
  -<src/lcd/menu/game/maze.cpp> -<src/lcd/menu/game/snake.cpp>
  -<src/lcd/menu/menu_backlash.cpp>
  -<src/lcd/menu/menu_bed_corners.cpp>
  -<src/lcd/menu/menu_bed_leveling.cpp>
  -<src/lcd/menu/menu_cancelobject.cpp>
  -<src/lcd/menu/menu_delta_calibrate.cpp>
  -<src/lcd/menu/menu_filament.cpp>
[... 197 lines cut ...]
  -<src/module/stepper/TMC26X.cpp>

#
# Default values apply to all 'env:' prefixed environments
#
[env]
framework         = arduino
extra_scripts     = ${common.extra_scripts}
build_flags       = ${common.build_flags}
lib_deps          = ${common.lib_deps}
monitor_speed     = 250000
monitor_eol       = LF
monitor_echo      = yes
monitor_filters   = colorize, time, send_on_enter

#
# Just print the dependency tree
#
[env:include_tree]
platform         = atmelavr
board            = megaatmega2560
build_flags      = -c -H -std=gnu++11 -Wall -Os -D__MARLIN_FIRMWARE__
build_src_filter = +<src/MarlinCore.cpp>

```

ivankravets, 2024-01-28:
> want to install packages for all the environments (20+)

Wow!!!!!!!! So, we have spent 10 years developing PlatformIO and letting developers bootstrap environments on demand to save networking resources whereas you try to do the opposite. So, why do you need 20+ envs and a dozen of gigabytes data?

artyomsv, 2024-01-28:
> > want to install packages for all the environments (20+)
> 
> Wow!!!!!!!! So, we have spent 10 years developing PlatformIO and letting developers bootstrap environments on demand to save networking resources whereas you try to do the opposite. So, why do you need 20+ envs and a dozen of gigabytes data?

The idea is to build a docker image that will have all the necessary dependencies. I do not know yet how big that image is going to be.

ivankravets, 2024-01-28:
It will not work. We update development platforms frequently including their dependencies. Each version of the development platform depends on the specified list of packages. For Marlin, you will need ~20Gb of storage. This is exactly what you should not do and PlatformIO was developed to solve one of these problems with package distribtuion.