platformio/platformio-core #4828: Release 6.1.12 breaks installation through github actions

staur1488, 2024-01-10:
I can't build expressLRS configuration

gives an error, here is its log

Сборка прошивки
Журналы
Failed to find Platformio on your computer. Trying to install it automatically...
Installer version: 1.1.2

Platform: Windows-10
Python version: 3.10.8 (tags/v3.10.8:aaaf517, Oct 11 2022, 16:50:30) [MSC v.1933 64 bit (AMD64)]
Python path: D:\����� �����\ExpressLRS Configurator\dependencies\windows_amd64\python\python3.exe

Creating a virtual environment at C:\Users\Igor\.platformio\penv

Updating Python package manager (PIP) in a virtual environment

PIP has been successfully updated!
Virtual environment has been successfully created!
Installing PlatformIO Core

Collecting platformio

  Using cached platformio-6.1.12-py3-none-any.whl

Collecting bottle==0.12.* (from platformio)

  Using cached bottle-0.12.25-py3-none-any.whl (90 kB)

Collecting click<9,>=8.0.4 (from platformio)

  Using cached click-8.1.7-py3-none-any.whl.metadata (3.0 kB)

Collecting colorama (from platformio)

  Using cached colorama-0.4.6-py2.py3-none-any.whl (25 kB)

Collecting marshmallow==3.* (from platformio)

  Using cached marshmallow-3.20.2-py3-none-any.whl.metadata (7.5 kB)

Collecting pyelftools==0.30 (from platformio)

  Using cached pyelftools-0.30-py2.py3-none-any.whl.metadata (381 bytes)

Collecting pyserial==3.5.* (from platformio)

  Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)

Collecting requests==2.* (from platformio)

  Using cached requests-2.31.0-py3-none-any.whl.metadata (4.6 kB)

Collecting semantic-version==2.10.* (from platformio)

  Using cached semantic_version-2.10.0-py2.py3-none-any.whl (15 kB)

Collecting tabulate==0.* (from platformio)

  Using cached tabulate-0.9.0-py3-none-any.whl (35 kB)

Collecting ajsonrpc==1.2.* (from platformio)

  Using cached ajsonrpc-1.2.0-py3-none-any.whl (22 kB)

Collecting starlette<0.35,>=0.19 (from platformio)

  Using cached starlette-0.34.0-py3-none-any.whl.metadata (5.8 kB)

Collecting uvicorn<0.26,>=0.16 (from platformio)
[... 136 lines cut ...]
    value = self.get(section, option)
  File "C:\Users\Igor\.platformio\penv\lib\site-packages\platformio\project\config.py", line 380, in get
    value = self.getraw(section, option, default)
  File "C:\Users\Igor\.platformio\penv\lib\site-packages\platformio\project\config.py", line 281, in getraw
    option_meta = self.find_option_meta(section, option)
  File "C:\Users\Igor\.platformio\penv\lib\site-packages\platformio\project\config.py", line 252, in find_option_meta
    scope = self.get_section_scope(section)
  File "C:\Users\Igor\.platformio\penv\lib\site-packages\platformio\project\config.py", line 167, in get_section_scope
    return section.split(":", 1)[0] if ":" in section else section
TypeError: argument of type 'NoneType' is not iterable

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `python -m pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

kyle-pennington-eroad, 2024-01-11:
Also having this issue locally on MacOS and on cloud instance of Ubuntu Linux on 6.1.12. Fails to parse `platformio.ini`.

ivankravets, 2024-01-11:
Could you share your `platformio.ini`?

staur1488, 2024-01-11:
[.platformio.zip](https://github.com/platformio/platformio-core/files/13898966/default.platformio.zip)
these are the files that the expressLRS configurator downloads

j123b567, 2024-01-11:
Minimal platformio.ini triggering this bug for us is

```ini
[env:mydev]
build_flags =
  -DHTTPD_FSDATA_FILE="\"${BUILD_DIR}/fsdata_custom.h\""
```

AndriySvs, 2024-01-11:
I have the same issue with expressLRS configuration as staur1488 reported.
But the interesting thing is that day before it was working fine.
So any suggestions how to resolve it?

staur1488, 2024-01-11:
I haven't found it yet.  :(

ivankravets, 2024-01-11:
Could you replace `${BUILD_DIR}` with `$BUILD_DIR`? See valid usage of [Interpolation of Values](https://docs.platformio.org/en/latest/projectconf/interpolation.html)

ivankravets, 2024-01-11:
@staur1488 , звідки береться `platformio.ini`? Його інший софт генерує?

staur1488, 2024-01-11:
@ivankravets його завантажте ExpressLRS конфігуратор. Пару днів тому все працювало. Не міняв нічого. Вчора з'явилась ця помилка

staur1488, 2024-01-11:
@ivankravets можу в десь в іншому застосунку скинути відео пояснення шо як робив. Бо в програмних діях не дуже розуміюсь

j123b567, 2024-01-11:
> Could you replace `${BUILD_DIR}` with `$BUILD_DIR`? See valid usage of [Interpolation of Values](https://docs.platformio.org/en/latest/projectconf/interpolation.html)

`$BUILD_DIR` does not solve the problem.
Replacing by `${sysenv.BUILD_DIR}` solves the problem.

So the system environment was accidentaly available in the `platformio.ini` namespace prior 6.1.12 and now it is not?
Maybe some more variables should be added as "built-in variables" to a global namespace to lower the impact of this breaking change.

Unfortunately, I was confused by a lot of other documentation (not about `platformio.ini`) where this works, and until 6.1.11, it works also in `platformio.ini`.
 - https://docs.platformio.org/en/latest/scripting/actions.html
 - https://docs.platformio.org/en/latest/scripting/examples/custom_program_name.html
 - etc.

It would be also great to just say `unknown value of $BUILD_DIR` instead of `TypeError: argument of type 'NoneType' is not iterable`

j123b567, 2024-01-11:
Others are probably using expressLRS and it has something like this in its platformio.ini
```ini
[env:ESP_TX_Backpack_via_PASSTHRU]
upload_command = python "$PROJECT_DIR/python/external/esptool/esptool.py" --passthrough -b $UPLOAD_SPEED ${UPLOAD_PORT and "-p "+UPLOAD_PORT} -c esp8266 --before passthru --after hard_reset write_flash 0x0000 "$SOURCE"
```

ivankravets, 2024-01-11:
> ${UPLOAD_PORT and "-p "+UPLOAD_PORT} 

Have you seen any similar examples for interpolation in our docs? The right solution for this case is https://docs.platformio.org/en/latest/scripting/index.html

The `$SCONS_VARIABLE` is valid syntax. Some public variables are documented in https://docs.platformio.org/en/latest/projectconf/sections/env/options/build/build_flags.html#built-in-variables

Nevertheless, we are investigating this issue.

j123b567, 2024-01-11:
> > ${UPLOAD_PORT and "-p "+UPLOAD_PORT}
> 
> Have you seen any similar examples for interpolation in our docs?

Never seen this syntax in your docs. Honestly, I don't understand, what it should do. I also don't know, what expressLRS really is. I only extracted the suspicious part of their `platformio.ini` file.

> Nevertheless, we are investigating this issue.

Great!

oturmanenko, 2024-01-11:
Faced the same problem.

ivankravets, 2024-01-11:
Could you re-test with `pio upgrade`? The issue should be fixed.

CapnBry, 2024-01-11:
There's no 6.1.13 available yet, but just editing my source with adab425 resolves the issue in both instances:
* `${PROJECTSRC_DIR}`
* `${UPLOAD_PORT and "-p "+UPLOAD_PORT}`

j123b567, 2024-01-11:
After `platformio upgrade --dev`
Unfortunately, it still breaks something like

```ini
[env:myenv]
platform = ststm32
board = genericSTM32F407VET6
build_flags =
  -Wl,-Map,${BUILD_DIR}/${PROGNAME}.map
```

With new error
```
 AttributeError `'str' object has no attribute 'map'' trying to evaluate `$PROGNAME.map'
```
This was working in 6.1.11

ivankravets, 2024-01-11:
@j123b567, thanks for helping with debugging this issue. There is yet another fix.

Please re-run `pio upgrade --dev`.

For people who use ExpressLRS, the PlatformIO Core should be located in the `%HOME_DIR%/.platformio/penv` folder. Unix - `bin/pio`, Windows - `Scripts/pio.exe`. 

Please confirm that the latest PlatformIO Core development branch compiles firmware without any modifications to the ExpressLRS firmware.

closed 2024-01-11 by ivankravets

mha1, 2024-01-11:
Updated via `python -m pip install -U https://github.com/platformio/platformio-core/archive/develop.zip`

finished ok I guess
```
...
WARNING: Ignoring invalid distribution ~ (C:\Users\10137\.platformio\penv\Lib\site-packages)
Installing collected packages: platformio
Successfully installed platformio-6.1.13a2
```

Built unmodified 3.x.x-maintenance and current main with no issues using VSCode/PlatformIO. I also built firmware with the ExpressLRS Configurator using GIT COMMIT to force it to discard pre-built cloud builds. All went well.

CapnBry, 2024-01-11:
I can also confirm ExpressLRS main firmware and backpack projects build again in VSCode as well as our Configurator with no changes.

I also had to update with `python -m pip install -U https://github.com/platformio/platformio-core/archive/develop.zip` as `pio upgrade --dev` reported Access Denied trying to update pio.exe.

j123b567, 2024-01-11:
> @j123b567, thanks for helping with debugging this issue. There is yet another fix.
Thank you. Confirming, that it works now for us.

kyle-pennington-eroad, 2024-01-11:
Thanks @ivankravets. Fixed on dev build for me now too.

mha1, 2024-01-11:
Thank you for your quick response

russelljahn, 2024-01-12:
Thanks for the fix!

I wasn't using ExpressLRS, but still ran into the same `TypeError: argument of type 'NoneType' is not iterable` issue with an unchanged project that had no issues a month ago.

Switching to the PIO dev branch fixed it for me.

ivankravets, 2024-01-12:
6.1.13 is out! https://github.com/platformio/platformio-core/releases/tag/v6.1.13

Thanks again for your help with this issue 🙏

commit adab425c (2024-01-11): Expanded support for SCons variables declared in the legacy format ``${SCONS_VARNAME}`` // Resolve #4828
commit f31f9fa6 (2024-01-11): Expanded support for SCons variables declared in the legacy format ``${SCONS_VARNAME}`` // Resolve #4828