platformio/platformio-core #4829: Platformio-ide not found, unable to build, NoneType error

MasculiousAdrenalus, 2024-01-12:
#### ISSUE:

Can't build or ```pio run``` 
command 'platformio-ide.serialMonitor' not found

------------------------------------------------------------------

### Configuration

Win10

--------------------------  ------------------------------------------------------
PlatformIO Core             6.1.12
Python                      3.12.1-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\<user>\.platformio
PlatformIO Core Executable  C:\Users\<user>\.platformio\penv\Scripts\platformio.exe
Python Executable           C:\Users\<user>\.platformio\penv\Scripts\python.exe
Global Libraries            0
Development Platforms       0
Tools & Toolchains          1
--------------------------  ------------------------------------------------------

### Actual Results
TypeError: Traceback (most recent call last):
  File "C:\Users\<user>\.platformio\penv\Lib\site-packages\platformio\__main__.py", line 103, in main
    cli()  # pylint: disable=no-value-for-parameter
    ^^^^^
  File "C:\Users\<user>\.platformio\penv\Lib\site-packages\click\core.py", line 1157, in __call__    
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\<user>\.platformio\penv\Lib\site-packages\click\core.py", line 1078, in main        
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Users\<user>\.platformio\penv\Lib\site-packages\platformio\cli.py", line 85, in invoke
    return super().invoke(ctx)
           ^^^^^^^^^^^^^^^^^^^
  File "C:\Users\<user>\.platformio\penv\Lib\site-packages\click\core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\<user>\.platformio\penv\Lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\<user>\.platformio\penv\Lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\<user>\.platformio\penv\Lib\site-packages\click\decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\<user>\.platformio\penv\Lib\site-packages\platformio\run\cli.py", line 145, in cli
    process_env(
  File "C:\Users\<user>\.platformio\penv\Lib\site-packages\platformio\run\cli.py", line 191, in process_env
    print_processing_header(name, config, verbose)
  File "C:\Users\<user>\.platformio\penv\Lib\site-packages\platformio\run\cli.py", line 232, in print_processing_header
    for k, v in config.items(env=env):
                ^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\<user>\.platformio\penv\Lib\site-packages\platformio\project\config.py", line 222, in items
    return [(option, self.get(section, option)) for option in self.options(section)]
                     ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\<user>\.platformio\penv\Lib\site-packages\platformio\project\config.py", line 380, in get
    value = self.getraw(section, option, default)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\<user>\.platformio\penv\Lib\site-packages\platformio\project\config.py", line 312, in getraw
    return self._expand_interpolations(section, option, value)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\<user>\.platformio\penv\Lib\site-packages\platformio\project\config.py", line 338, in _expand_interpolations
    return self.VARTPL_RE.sub(
           ^^^^^^^^^^^^^^^^^^^
  File "C:\Users\<user>\.platformio\penv\Lib\site-packages\platformio\project\config.py", line 339, in <lambda>
    lambda match: self._re_interpolation_handler(section, option, match), value
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\<user>\.platformio\penv\Lib\site-packages\platformio\project\config.py", line 367, in _re_interpolation_handler
    value = self.get(section, option)
            ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\<user>\.platformio\penv\Lib\site-packages\platformio\project\config.py", line 380, in get
    value = self.getraw(section, option, default)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\<user>\.platformio\penv\Lib\site-packages\platformio\project\config.py", line 281, in getraw
    option_meta = self.find_option_meta(section, option)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\<user>\.platformio\penv\Lib\site-packages\platformio\project\config.py", line 252, in find_option_meta
    scope = self.get_section_scope(section)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\<user>\.platformio\penv\Lib\site-packages\platformio\project\config.py", line 167, in get_section_scope
    return section.split(":", 1)[0] if ":" in section else section
                                       ^^^^^^^^^^^^^^
TypeError: argument of type 'NoneType' is not iterable

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `python -m pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

### Expected Results
It build

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini

[platformio]
default_envs = main

[env]
platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.5.3/platform-espressif32-2.0.5.3.zip

build_type = debug

;; ############################################################
[env:main]
framework = arduino
board = esp-wrover-kit
```

russelljahn, 2024-01-12:
This might be related to https://github.com/platformio/platformio-core/issues/4828.

I ran into the same issue where PIO threw a `TypeError: argument of type 'NoneType' is not iterable` exception when parsing my `platformio.ini` file, and reported it was invalid. 

I switched to the dev branch like the attached issue (run `pio upgrade --dev`), and using the dev branch fixed it for me.

MasculiousAdrenalus, 2024-01-12:
Thank you so much, yes that fixed it.

closed 2024-01-12 by MasculiousAdrenalus