platformio/platformio-core #4833: Bad CPU type in executable clang-tidy on Apple M3 Pro

jonathanfoster, 2024-01-13:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to the
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues) repository

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

- macOS Sonoma 14.2.1
- Darwin Kernel Version 23.2.0: Wed Nov 15 21:54:51 PST 2023; root:xnu-10002.61.3~2/RELEASE_ARM64_T6030
- Apple M3 Pro

**PlatformIO Version** (`platformio --version`):

- PlatformIO Core, version 6.1.13

### Description of problem

When running `pio check` with `clangtidy` on an Apple M3 Pro, the following error is returned:

```bash
OSError: [Errno 86] Bad CPU type in executable: '/Users/jonathanfoster/.platformio/packages/tool-clangtidy/clang-tidy'
```

It appears the incorrect architecture is choosen when installing `tool-clangtidy`.

```bash
$ arch
arm64
$ lipo -info /Users/jonathanfoster/.platformio/packages/tool-clangtidy/clang-tidy
Non-fat file: /Users/jonathanfoster/.platformio/packages/tool-clangtidy/clang-tidy is architecture: x86_64
```

Uninstalling and reinstalling `tool-clandtidy` produces the same error.

#### Steps to Reproduce

1. Configure `platformio.ini` to use clang-tidy.

      ```ini
      check_tool = clangtidy
      ```

2. Run static code analysis.

      ```bash
      pio check
      ```

### Actual Results

```bash

$ pio check
Checking lilygo-t-display-s3 > clangtidy (board: lilygo-t-display-s3; platform: espressif32; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
OSError: Traceback (most recent call last):
  File "/opt/homebrew/Cellar/platformio/6.1.13/libexec/lib/python3.12/site-packages/platformio/__main__.py", line 103, in main
    cli()  # pylint: disable=no-value-for-parameter
    ^^^^^
  File "/opt/homebrew/opt/python-click/lib/python3.12/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/opt/python-click/lib/python3.12/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/opt/homebrew/Cellar/platformio/6.1.13/libexec/lib/python3.12/site-packages/platformio/cli.py", line 85, in invoke
    return super().invoke(ctx)
           ^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/opt/python-click/lib/python3.12/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/opt/python-click/lib/python3.12/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/opt/python-click/lib/python3.12/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/Cellar/platformio/6.1.13/libexec/lib/python3.12/site-packages/platformio/check/cli.py", line 144, in cli
    rc = ct.check(
         ^^^^^^^^^
  File "/opt/homebrew/Cellar/platformio/6.1.13/libexec/lib/python3.12/site-packages/platformio/check/tools/base.py", line 239, in check
    self.execute_check_cmd(cmd)
  File "/opt/homebrew/Cellar/platformio/6.1.13/libexec/lib/python3.12/site-packages/platformio/check/tools/base.py", line 192, in execute_check_cmd
    result = proc.exec_command(
             ^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/Cellar/platformio/6.1.13/libexec/lib/python3.12/site-packages/platformio/proc.py", line 114, in exec_command
    with subprocess.Popen(*args, **kwargs) as p:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/Cellar/python@3.12/3.12.1/Frameworks/Python.framework/Versions/3.12/lib/python3.12/subprocess.py", line 1026, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/opt/homebrew/Cellar/python@3.12/3.12.1/Frameworks/Python.framework/Versions/3.12/lib/python3.12/subprocess.py", line 1950, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
OSError: [Errno 86] Bad CPU type in executable: '/Users/jonathanfoster/.platformio/packages/tool-clangtidy/clang-tidy'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `python -m pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

### Expected Results

`pio check` should run without error.

jonathanfoster, 2024-01-13:
Looks like the issue is with the `tool-clangtidy` registry config, the files for `darwin_amd46` appear to be misconfigured. `darwin_arm64 ` should be a different download URL than `darwin_x86_64`.

```json
"files": [
  {
    "name": "tool-clangtidy-darwin_x86_64-1.150005.0.tar.gz",
    "size": 19220346,
    "checksum": {
      "sha256": "dd2b0f982558194a4d0bda3fd15e084af42fabe495aac7805e458b496ea6587d"
    },
    "system": [
      "darwin_x86_64",
      "darwin_arm64"
    ],
    "download_url": "https://dl.registry.platformio.org/download/platformio/tool/tool-clangtidy/1.150005.0/tool-clangtidy-darwin_x86_64-1.150005.0.tar.gz"
  }
]
```

ivankravets, 2024-01-13:
There is no native binary for Mac ARM. 

Do you have Rosetta installed?

jonathanfoster, 2024-01-13:
@ivankravets It looks like LLVM does have [native binaries for Mac ARM now](https://github.com/llvm/llvm-project/releases/download/llvmorg-17.0.6/clang+llvm-17.0.6-arm64-apple-darwin22.0.tar.xz). 

I can install Rosetta or even copy an arm64 binary from LLVM's repo as a work around, but it would be a better experience if this was supported in the registry. Any idea what's required to update the package config and provide an arm64 download URL?

ivankravets, 2024-01-13:
I agree with you. There is another problem. 95% of all toolchains are NOT native ARM. You will still need Rosetta.

jonathanfoster, 2024-01-13:
In my case, I'm using the espressif toolchain and clang-tidy is the only tool that isn't native ARM. Would it make sense to update packages incrementally as arm64 is supported? What's required to update the `tool-clangtidy` package config and provide an arm64 download URL?