platformio/platformio-core #4838: LDF can't work with Adafruit TinyUSB on any setting

maxgerhardt, 2024-01-23:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 10 x64

**PlatformIO Version** (`platformio --version`): `PlatformIO Core, version 6.1.13`

### Description of problem

A Atmel SAM project that attempts to include Adafruit TinyUSB library of the latest version (2.3.2 down to and including 2.2.2)  fails to properly compile, not finding the `SPI.h` header declared in the `Adafruit_USBH_Host.h` file. Version 2.2.1 compiles fine, because it did not have an include for SPI.h.

The LDF fails to find the dependency on SPI.h no matter the `lib_ldf_mode` setting.

#### Steps to Reproduce

1. Use `platformio.ini` and `src/main.cpp` provided below
2. Build
3. Observe build failure
4. Change version to `adafruit/Adafruit TinyUSB Library@2.2.1` and rebuild
5. Oberve compile fine

### Actual Results

```
Compiling .pio\build\busylight\lib4ff\Adafruit_ZeroDMA\Adafruit_ZeroDMA.cpp.o
Compiling .pio\build\busylight\lib361\SPI\SPI.cpp.o
In file included from .pio\libdeps\busylight\Adafruit TinyUSB Library\src/Adafruit_TinyUSB.h:76,
                 from C:\Users\Max\.platformio\packages\framework-arduino-samd-adafruit\libraries\Adafruit_ZeroDMA\Adafruit_ZeroDMA.cpp:35:
.pio\libdeps\busylight\Adafruit TinyUSB Library\src/arduino/Adafruit_USBH_Host.h:30:10: fatal error: SPI.h: No such file or directory
```

### Expected Results

Compiles fine

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:busylight]
platform = atmelsam
board = adafruit_neotrinkey_m0
framework = arduino
lib_deps=
    adafruit/Adafruit TinyUSB Library@2.3.2
lib_archive = no
build_flags=
    -DUSE_TINYUSB
lib_ldf_mode = deep+
lib_ignore = USBHost
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>
#include "Adafruit_TinyUSB.h"

void setup() {}
void loop() {}
```

### Additional info

Especially concerning are the messages in  verbose mode that the LDF fail to evalute the TinyUSB macro `TU_CHECK_MCU` correctly.

>Warning! Classic Pre Processor is used for `C:\Users\Max\temp\trinkey\.pio\libdeps\busylight\Adafruit TinyUSB Library\src\arduino\Adafruit_TinyUSB_API.cpp`, advanced has failed with `Incorrect number of arguments to `TU_CHECK_MCU'`
Warning! Classic Pre Processor is used for `C:\Users\Max\temp\trinkey\.pio\libdeps\busylight\Adafruit TinyUSB Library\src\arduino\Adafruit_USBD_CDC.cpp`, advanced has failed with `Incorrect number of arguments to `TU_CHECK_MCU'`
Warning! Classic Pre Processor is used for `C:\Users\Max\temp\trinkey\.pio\libdeps\busylight\Adafruit TinyUSB Library\src\arduino\Adafruit_USBD_Device.cpp`, advanced has failed with `Incorrect number of arguments to `TU_CHECK_MCU'`
Warning! Classic Pre Processor is used for `C:\Users\Max\temp\trinkey\.pio\libdeps\busylight\Adafruit TinyUSB Library\src\arduino\Adafruit_USBH_Host.cpp`, advanced has failed with `Incorrect number of arguments to `TU_CHECK_MCU'`

valeros, 2024-02-07:
Hi @maxgerhardt, thanks for reporting. I've reproduced the issue and it seems it's caused by a circular dependency between `Adafruit TinyUSB`, `Adafruit Zero DMA` and `SPI` libraries. I'm wondering if the latest versions of the `TinyUSB` library are compatible with their SAMD core. The libraries in the SAMD repo were [updated](https://github.com/adafruit/ArduinoCore-samd/commits/master/libraries) just yesterday and they still prefer to stick to a quite old version `1.4.7` of the `Adafruit TinyUSB` library for some reason.

As a workaround I'd recommend using the library version bundled with the framework package at the moment.

nerdyscout, 2024-06-14:
I think this relates to my issue here: https://community.platformio.org/t/outdated-framework-arduino-samd-adafruit/41153