platformio/platformio-core #4839: Feature request to support python extra_scripts with dependency manager

elon-code, 2024-01-24:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to the
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues) repository

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [x] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

**PlatformIO Version** (`platformio --version`): 6.1.13

### Description of request
As of now, I believe you can include extra_scripts repository link in the lib_deps flags, however, it would be nice if extra scripts were more fully supported in the dependency manager. I think these features would be great to add:
- Have a separate section for extra_scripts dependencies, possibly with the ability to just paste repo links into the extra_scripts section as is.
- Download extra_scripts into a separate folder from normal libdeps
- Ability to call these scripts in an easy way. Just using the relative path would be fine, but it might be useful to have it so I can call these scripts automatically when included in dependency manager.

Thanks for your help and for the work you do! I may be able to contribute to helping develop this feature if it is desired, let me know if there is anything I can do.

ivankravets, 2024-01-26:
Do you mean https://docs.platformio.org/en/latest/manifests/library-json/fields/build/extrascript.html ?

elon-code, 2024-01-27:
Yes, this works! I do kind of think it might be best to have a differentiation between scripts and libraries in the dependency manager and registry... but it works good enough as is.

As it works well enough for my use case, feel free to close at your discretion. Thank you for your help and for your support of PlatformIO!

ivankravets, 2024-01-27:
Happy coding with PlatformIO! 🚀

closed 2024-01-27 by ivankravets