platformio/platformio-core #4847: Problems with –project-dir option to pio init

NormanDunbar, 2024-01-31:
What kind of issue is this?

- [ *] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.
------------------------------------------------------------------

### Configuration

**Operating system**: Linux Mint 21.1 Vera (64 bit.)

**PlatformIO Version** : 6.1.13

### Description of problem

I'm experimenting because I'm working through the author review part of the second edition of *Arduino Software Internals*, and the PlatformIO chapter has been updated.

I note that there is a `--project-dir` or `-d` option to `pio init`. However, the operation is inconsitent.

In my first test, I didn't create the output directory, and it failed, advising me that the output directory didn't exist. I can cope with that, but it would be nice if it created the output as it creates the `src`, `lib` directories anyway, normally.

In my second test, I created the directory and specified it thus:

```
mkdir fred
pio init -b uno -d fred
```

That fell over with a lot of red text and Python errors---see below. However, it did create the usual directories and *most* of `platformio.ini`---the comments were there but the `[env:uno]` environment was missing.

I cleared out and tried again with a relative path.

```
rm -Rf fred
mkdir fred
pio init -b uno -d ./fred
```

Same problem, same messages, same output.

Finally, I tested with a full path:

```
rm -Rf fred
mkdir fred
pio init -b uno -d $PWD/fred
```

And this worked perfectly.

#### Steps to Reproduce

1. Create a new directory, `xxxx`.
2. Run `pio init -b uno -d xxxx` or `pio init -b uno -d ./xxxx` and it will fail.
3. Run `pio init -b uno -d $PWD/xxxx` and it will work.

### Actual Results

The output from the failing tests is:

```
The following files/directories have been created in fred
include - Put project header files here
lib - Put project specific (private) libraries here
src - Put project source files here
platformio.ini - Project Configuration File
FileNotFoundError: Traceback (most recent call last):
  File "/home/norman/.platformio/penv/lib/python3.10/site-packages/platformio/__main__.py", line 103, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/home/norman/.platformio/penv/lib/python3.10/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/home/norman/.platformio/penv/lib/python3.10/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/home/norman/.platformio/penv/lib/python3.10/site-packages/platformio/cli.py", line 85, in invoke
    return super().invoke(ctx)
  File "/home/norman/.platformio/penv/lib/python3.10/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/norman/.platformio/penv/lib/python3.10/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/norman/.platformio/penv/lib/python3.10/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/home/norman/.platformio/penv/lib/python3.10/site-packages/platformio/project/commands/init.py", line 92, in project_init_cmd
    update_board_envs(project_dir, boards, project_options, env_prefix)
  File "/home/norman/.platformio/penv/lib/python3.10/site-packages/platformio/project/commands/init.py", line 336, in update_board_envs
    config.save()
  File "/home/norman/.platformio/penv/lib/python3.10/site-packages/platformio/project/config.py", line 551, in save
    with open(path or self.path, mode="w+", encoding="utf8") as fp:
FileNotFoundError: [Errno 2] No such file or directory: 'fred/platformio.ini'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `python -m pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

### Expected Results

I expected the `platformio.ini` file to be correctly created with the environment for an Uno board.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

```

**Source file to reproduce issue:**

No source file as the `pio init` command failed.

### Additional info

The bit about "FileNotFoundError: [Errno 2] No such file or directory: 'fred/platformio.ini'" seems misleading as it managed to create the file, although I suspect that's more of a "copy and paste" operation. I suspect it's trying to open the file to write the environment settings. I'm not a Python programmer but it looks to me like it can't tell the difference between a full path or a relative one, and I find that hard to believe.

valeros, 2024-02-02:
Hi @NormanDunbar, thanks for reporting. Indeed seems like a bug to me, although it only occurs when the `--board` argument is set.

PS: it's highly desirable to use the actual "[pio project init](https://docs.platformio.org/en/latest/core/userguide/project/cmd_init.html)" command for initializing new projects.

ivankravets, 2024-02-02:
Does `pio project init -d` work?

NormanDunbar, 2024-02-05:
Sorry about the delay getting back to you.
@valeros I have not been aware of `pio project init` until you mentioned it, thanks. I'll try to use it in future, however, the same bug behaviour is seen if the board option is specified.

@ivankravets The project will be initialised if I specify just the `-d xxx` option, provided 'xxx' is an existing directory. If I use the `-b` option, I get the same errors as with `pio init -b uno -d xxx`.

Cheers,
Norman.

closed 2024-02-07 by ivankravets

NormanDunbar, 2024-02-10:
Thank you both.

Regards,
Norman.

commit 353f4403 (2024-02-07): Addressed an issue where passing a relative path to the `pio project init` // Resolve #4847