platformio/platformio-core #4849: Library Dependency Finder (LDF) adds too many files when testing (pio test)

dhebbeker, 2024-02-01:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

* **Operating system**: GNU/Linux
* **PlatformIO Version** (`platformio --version`): PlatformIO Core, version 6.1.14a1

### Description of problem

When testing (`pio test`) the LDF adds too many files to the build configuration: Source files which are in the same directory as a dependency are compiled, even when 

* there is no `#include` chain and
* the name of the source file does not have the same name as the header file and
* `lib_ldf_mode = chain`

For testing I modified the [calculator example project](https://github.com/platformio/platformio-examples/tree/f06e9656a490c17b9193ce78dca8df4c7a6cb82e/unit-testing/calculator) by adding (https://github.com/platformio/platformio-examples/commit/d98e5a746f12c4692afaefba9b925869e96a4ce7) the file `extra.cpp` (see below):

```plain
├── lib
│   ├── calculator
│   │   └── src
│   │       ├── calculator.cpp
│   │       ├── calculator.h
│   │       └── extra.cpp         ← I added this file
│   └── README
├── platformio.ini
├── README.rst
├── src
│   └── main.cpp
└── test
    ├── README
    ├── test_common
    │   └── test_calculator.cpp
    ├── test_desktop
    │   └── test_calculator.cpp
    └── test_embedded
        └── test_calculator.cpp
```

This file has no `#include` statement, nor is its name the same as any header included.

From the [documentation of LDF mode](https://docs.platformio.org/en/latest/librarymanager/ldf.html#dependency-finder-mode) `chain`:

> Parses ALL C/C++ source files of the project and follows only by nested includes (`#include ...`, chain...) from the libraries. It also parses C, CC, CPP files from libraries which have the same name as included header file.

I do not see why this file should be built.

Also in the documentation the example ("Case 2") indicates that `extra.cpp` is not "known".

#### Steps to Reproduce

1. Checkout [the modified calculator project](https://github.com/dhebbeker/platformio-examples/commit/d98e5a746f12c4692afaefba9b925869e96a4ce7) or simply add an unrelated source file to a `lib/` dependency which will fail to compile.
2. Assert that the LDF mode is `chain`.
3. Run the test: `platformio test --environment native -vvv`

### Actual Results

```bash
platformio-examples/unit-testing/calculator$ platformio test --environment native -vvv
Collected 3 tests (test_common, test_desktop, test_embedded)

Processing test_desktop in native environment
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Building...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 2 compatible libraries
Scanning dependencies...
Dependency Graph
|-- calculator (License: Unknown, Path: ~/pro/platformio-examples/unit-testing/calculator/lib/calculator)
|-- Unity @ 2.5.2 (License: MIT, Path: ~/pro/platformio-examples/unit-testing/calculator/.pio/libdeps/native/Unity)
Building in test mode
g++ -o .pio/build/native/lib4d4/calculator/extra.o -c -DPLATFORMIO=60114 -DPIO_UNIT_TESTING -Ilib/calculator/src lib/calculator/src/extra.cpp
lib/calculator/src/extra.cpp:1:15: error: static assertion failed
    1 | static_assert(false);
      |               ^~~~~
*** [.pio/build/native/lib4d4/calculator/extra.o] Error 1
Building stage has failed, see errors above. Use `pio test -vvv` option to enable verbose output.
--------------------------------------------------------------------------------------------------- native:test_desktop [ERRORED] Took 0.39 seconds ---------------------------------------------------------------------------------------------------
```

### Expected Results

Compilation should succeed and test should run (with expected failure), as `extra.cpp` is ignored.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter, extra scripting
;   Upload options: custom port, speed and extra flags
;   Library options: dependencies, extra library storages
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

;
; Embedded platforms
;

[env:uno]
platform = atmelavr
framework = arduino
board = uno
test_ignore = test_desktop

;
; Desktop platforms (Win, Mac, Linux, Raspberry Pi, etc)
; See https://platformio.org/platforms/native
;

[env:native]
platform = native
test_ignore = test_embedded

```

**Source file to reproduce issue:**

`extra.cpp`:
```cpp
static_assert(false);

```

### Additional info

The MVP is based on the calculator example. I simply added a file to the library to confirm that this file is added to the build configuration. That is not what I expected from my understanding of the documentation.

I did search for posts in the community forums and similar issues in this GitHub project. I did not find anything similar.

From my point of view either the LDF does not work properly or its documentation should be adjusted or completed.

ivankravets, 2024-02-07:
Once the library is picked by LDF for the build process, the all files will be built. You can control this with https://docs.platformio.org/en/latest/manifests/library-json/fields/build/srcfilter.html or https://docs.platformio.org/en/latest/manifests/library-json/fields/build/extrascript.html

closed 2024-02-07 by ivankravets

dhebbeker, 2024-02-07:
I would appreciate if this

> Once a library is picked by LDF for the build process, all files will be built.

would be included to the LDF documentation.