platformio/platformio-core #4857: In the results of the test, different file names are displayed.

7-rate, 2024-02-10:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to the
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues) repository

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ x ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:
Windows 11

**PlatformIO Version** (`platformio --version`):
PlatformIO Core, version 6.1.13

### Description of problem
In the results of the test, different file names are displayed.

#### Steps to Reproduce

1.Write the following code in test_main.cpp
```c
void setup() {
    // NOTE!!! Wait for >2 secs
    // if board doesn't support software reset via Serial.DTR/RTS
    delay( 2000 );

    UNITY_BEGIN();
    RUN_TEST( test_sandbox );
    RUN_TEST( test_tm );
    RUN_TEST( test_dm );
    RUN_TEST( test_calc_utils );
    UNITY_END();
}
```
test_sandbox and other functions are in separate files, such as test_sandbox.cpp.

2.pio test 

### Actual Results
```
Building & Uploading...
Testing...
If you don't see any output for the first 10 secs, please reset board (press reset button)

test\test_main.cpp:24: test_sandbox     [PASSED]
test\test_main.cpp:25: test_tm  [PASSED]
test\test_main.cpp:26: test_dm  [PASSED]
test\test_main.cpp:77: test_fast_exp    [PASSED]
test\test_main.cpp:78: test_calc_custom_sigmoid [PASSED]
test\test_main.cpp:79: test_calc_custom_sigmoid_k_min   [PASSED]
test\test_main.cpp:80: test_calc_angle  [PASSED]
test\test_main.cpp:80: test_calc_angle  [PASSED]
```

### Expected Results
```
Building & Uploading...
Testing...
If you don't see any output for the first 10 secs, please reset board (press reset button)

test\test_sandbox.cpp:24: test_sandbox     [PASSED]
test\test_time_measure.cpp:25: test_tm  [PASSED]
:
:
:
```

ivankravets, 2024-02-10:
Could you provide a simple project as a ZIP archive and attach it to this issue? 

Also, have you read this https://docs.platformio.org/en/latest/advanced/unit-testing/structure/hierarchy.html ?

7-rate, 2024-02-11:
A simplified project is attached.
[test_proj.zip](https://github.com/platformio/platformio-core/files/14232135/test_proj.zip)

In this project, I dare to make ```test_time_measure.cpp:16``` to be ERROR.

The result is shown below.
```
test\test_main.cpp:21: test_sandbox     [PASSED]
test\test_main.cpp:16: test_tm: Values Not Within Delta 50 Expected 2000 Was 1045       [FAILED]
test\test_main.cpp:23: test_dm  [PASSED]
```

On the second line, the expected log is as follows
```
test\test_time_measure.cpp:16: test_tm: Values Not Within Delta 50 Expected 2000 Was 1045       [FAILED]
```

The URL is difficult for me to decipher. I will try to read more.

valeros, 2024-02-12:
Hi @7-rate, I believe that's expected behavior for the Unity test framework. The test locations are pulled from the file where they are run. In you case, despite the original location of the test functions, they were just implicitly imported into the `test_main.cpp` file where they are eventually executed by the test runner. If you want proper test names, then you need to organize them in a special way (more info on the [Test Hierarchy](https://docs.platformio.org/en/latest/advanced/unit-testing/structure/hierarchy.html#unit-testing-test-hierarchy)). For example the following structure:
```
test
|-- test_distance_measure
|   `-- test_distance_measure.cpp
|-- test_sandbox
|   `-- test_sandbox.cpp
`-- test_time_measure
    `-- test_time_measure.cpp
```
will have the following results:
[image]

closed 2024-02-12 by valeros