platformio/platformio-core #4858: NameError: name 'includes' is not defined while updating metadata for VSCode IDE

mbariola, 2024-02-11:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to the
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues) repository

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ *] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:
Win11

**PlatformIO Version** (`platformio --version`):
6.1.14a1

### Description of problem
I have opened an old platformio project folder after being prompted that platformio would be updated to 3.3.3.
After fixing $PROJECT_HASH into ${PROJECT_HASH}, while PlatformIO is updating metadata for VSCode IDE, the following is output to the terminal

Resolving wemos_d1_mini32 dependencies...
Already up-to-date.
Updating metadata for the vscode IDE...
NameError: Traceback (most recent call last):
  File "C:\Users\massi\.platformio\penv\Lib\site-packages\platformio\__main__.py", line 103, in main
    cli()  # pylint: disable=no-value-for-parameter
    ^^^^^
  File "C:\Users\massi\.platformio\penv\Lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\massi\.platformio\penv\Lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Users\massi\.platformio\penv\Lib\site-packages\platformio\cli.py", line 85, in invoke
    return super().invoke(ctx)
           ^^^^^^^^^^^^^^^^^^^
  File "C:\Users\massi\.platformio\penv\Lib\site-packages\click\core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\massi\.platformio\penv\Lib\site-packages\click\core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\massi\.platformio\penv\Lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\massi\.platformio\penv\Lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\massi\.platformio\penv\Lib\site-packages\platformio\project\commands\init.py", line 122, in project_init_cmd
    generator.generate()
  File "C:\Users\massi\.platformio\penv\Lib\site-packages\platformio\project\integration\generator.py", line 165, in generate
    contents = self._render_tpl(tpl_path, tpl_vars)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\massi\.platformio\penv\Lib\site-packages\platformio\project\integration\generator.py", line 171, in _render_tpl
    return bottle.template(fp.read(), **tpl_vars)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\massi\.platformio\penv\Lib\site-packages\bottle.py", line 3668, in template
    return TEMPLATES[tplid].render(kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\massi\.platformio\penv\Lib\site-packages\bottle.py", line 3457, in render
    self.execute(stdout, env)
  File "C:\Users\massi\.platformio\penv\Lib\site-packages\bottle.py", line 3444, in execute
    eval(self.co, env)
  File "<string>", line 75, in <module>
**NameError: name 'includes' is not defined**

#### Steps to Reproduce

1.
2.
3.

### Actual Results
as above

### Expected Results
no errors or exceptions

### Additional info

I followed what indicated in issue #4754 but it did not solve the issue

ivankravets, 2024-02-12:
Could you reproduce this issue after VSCode restart?

mbariola, 2024-02-12:
@ivankravets yes, I also moved everything to a different, simpler path thinking that maybe there were spaces or something there - but the issue is consistently happening

ivankravets, 2024-02-12:
Could you share your `platformio.ini`?

mbariola, 2024-02-13:
sure, here it is @ivankravets 

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
default_envs = wemos_d1_mini32
;workspace_dir = ${sysenv.TEMP}/pio-build/${PROJECT_HASH} // workspace_dir or build_dir do not change the result
build_dir = ${sysenv.TEMP}/pio-build/${PROJECT_HASH}

[env:wemos_d1_mini32]
platform = espressif32
board = wemos_d1_mini32
framework = arduino
monitor_speed = 460800
monitor_filters = esp32_exception_decoder, send_on_enter, colorize, time
lib_deps = 
	h2zero/NimBLE-Arduino@^1.3.3
	https://github.com/pschatzmann/arduino-audio-tools
	https://github.com/pschatzmann/ESP32-A2DP.git
	fastled/FastLED @ ^3.5.0
	https://github.com/me-no-dev/ESPAsyncWebServer.git
	ayushsharma82/AsyncElegantOTA@^2.2.7
	arduino-libraries/Arduino_JSON@^0.1.0
	marvinroger/AsyncMqttClient@^0.9.0
build_flags = -DCORE_DEBUG_LEVEL=2 -Wno-unused-variable -Wno-unused-but-set-variable -Wno-unused-function -Wno-format-extra-args
```

ivankravets, 2024-02-13:
Could you run in the project directory:

```
pio run -t __idedata -e wemos_d1_mini32
```

Do you see any errors?

mbariola, 2024-02-13:
no errors 

```
PS C:\Users\massi\Desktop\PlatformIO\ESP32_Gong> pio run -t __idedata -e wemos_d1_mini32
Processing wemos_d1_mini32 (platform: espressif32; board: wemos_d1_mini32; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/wemos_d1_mini32.html
PLATFORM: Espressif 32 (6.1.0) > WEMOS D1 MINI ESP32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20007.0 (2.0.7) 
 - tool-esptoolpy @ 1.40500.0 (4.5.0) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 56 compatible libraries
Scanning dependencies...
Dependency Graph
|-- NimBLE-Arduino @ 1.4.0
|-- audio-tools @ 0.8.0+sha.ec07518
|-- ESP32-A2DP @ 1.7.1+sha.6aaf08d
|-- FastLED @ 3.5.0
|-- ESP Async WebServer @ 1.2.3+sha.f71e3d4
|-- AsyncElegantOTA @ 2.2.7
|-- Arduino_JSON @ 0.1.0
|-- AsyncMqttClient @ 0.9.0
|-- AsyncTCP @ 1.1.1
|-- CapSense
|-- MyUtils
|-- Preferences @ 2.0.0
|-- quickPatterns @ 0.2
|-- SPIFFS @ 2.0.0
|-- VibSense
Building in release mode

{"build_type": "release", "env_name": "wemos_d1_mini32", "libsource_dirs": ["C:\\Users\\massi\\Desktop\\PlatformIO\\ESP32_Gong\\lib", "C:\\Users\\massi\\Desktop\\PlatformIO\\ESP32_Gong\\.pio\\libdeps\\wemos_d1_mini32", "C:\\Users\\massi\\.platformio\\lib", "C:\\Users\\massi\\.platformio\\packages\\framework-arduinoespressif32\\libraries"], "defines": ["PLATFORMIO=60114", "ARDUINO_D1_MINI32", "CORE_DEBUG_LEVEL=2", "HAVE_CONFIG_H", "MBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"", "UNITY_INCLUDE_CONFIG_H", "WITH_POSIX", "_GNU_SOURCE", "IDF_VER=\"v4.4.4\"", "ESP_PLATFORM", "_POSIX_READER_WRITER_LOCKS", "ARDUINO_ARCH_ESP32", "ESP32", "F_CPU=240000000L", "ARDUINO=10812", "ARDUINO_VARIANT=\"d1_mini32\"", "ARDUINO_BOARD=\"WEMOS D1 MINI ESP32\"", "ARDUINO_PARTITION_default"], "includes": {"build": ["C:\\Users\\massi\\Desktop\\PlatformIO\\ESP32_Gong\\include", "C:\\Users\\massi\\Desktop\\PlatformIO\\ESP32_Gong\\src", "C:\\Users\\massi\\Desktop\\PlatformIO\\ESP32_Gong\\lib\\VibSense", "C:\\Users\\massi\\.platformio\\packages\\framework-arduinoespressif32\\libraries\\SPIFFS\\src", "C:\\Users\\massi\\Desktop\\PlatformIO\\ESP32_Gong\\lib\\quickPatterns\\src", "C:\\Users\\massi\\Desktop\\PlatformIO\\ESP32_Gong\\lib\\CapSense", "C:\\Users\\massi\\.platformio\\packages\\framework-arduinoespressif32\\libraries\\Preferences\\src", "C:\\Users\\massi\\Desktop\\PlatformIO\\ESP32_Gong\\lib\\MyUtils", "C:\\Users\\massi\\Desktop\\PlatformIO\\ESP32_Gong\\.pio\\libdeps\\wemos_d1_mini32\\AsyncMqttClient\\src", "C:\\Users\\massi\\Desktop\\PlatformIO\\ESP32_Gong\\.pio\\libdeps\\wemos_d1_mini32\\Arduino_JSON\\src", "C:\\Users\\massi\\.platformio\\packages\\framework-arduinoespressif32\\libraries\\Update\\src", "C:\\Users\\massi\\Desktop\\PlatformIO\\ESP32_Gong\\.pio\\libdeps\\wemos_d1_mini32\\ESP Async WebServer\\src", "C:\\Users\\massi\\Desktop\\PlatformIO\\ESP32_Gong\\.pio\\libdeps\\wemos_d1_mini32\\AsyncElegantOTA\\src", "C:\\Users\\massi\\.platformio\\packages\\framework-arduinoespressif32\\libraries\\FS\\src", "C:\\Users\\massi\\Desktop\\PlatformIO\\ESP32_Gong\\.pio\\libdeps\\wemos_d1_mini32\\AsyncTCP\\src", "C:\\Users\\massi\\Desktop\\PlatformIO\\ESP32_Gong\\.pio\\libdeps\\wemos_d1_mini32\\FastLED\\src", "C:\\Users\\massi\\.platformio\\packages\\framework-arduinoespressif32\\libraries\\SPI\\src", "C:\\Users\\massi\\Desktop\\PlatformIO\\ESP32_Gong\\.pio\\libdeps\\wemos_d1_mini32\\ESP32-A2DP\\src", "C:\\Users\\massi\\Desktop\\PlatformIO\\ESP32_Gong\\.pio\\libdeps\\wemos_d1_mini32\\audio-tools\\src", "C:\\Users\\massi\\.platformio\\packages\\framework-arduinoespressif32\\libraries\\WiFiClientSecure\\src", "C:\\Users\\massi\\.platformio\\packages\\framework-arduinoespressif32\\libraries\\WiFi\\src", "C:\\Users\\massi\\Desktop\\PlatformIO\\ESP32_Gong\\.pio\\libdeps\\wemos_d1_mini32\\NimBLE-Arduino\\src", "C:\\Users\\massi\\.platformio\\packages\\framework-arduinoespressif32\\tools\\sdk\\esp32\\include\\newlib\\platform_include", "C:\\Users\\massi\\.platformio\\packages\\framework-arduinoespressif32\\tools\\sdk\\esp32\\include\\freertos\\include", "C:\\Users\\massi\\.platformio\\packages\\framework-arduinoespressif32\\tools\\sdk\\esp32\\include\\freertos\\include\\esp_additions\\freertos", "C:\\Users\\massi\\.platformio\\packages\\framework-arduinoespressif32\\tools\\sdk\\esp32\\include\\freertos\\port\\xtensa\\include", "C:\\Users\\massi\\.platformio\\packages\\framework-arduinoespressif32\\tools\\sdk\\esp32\\include\\freertos\\include\\esp_additions", "C:\\Users\\massi\\.platformio\\packages\\framework-arduinoespressif32\\tools\\sdk\\esp32\\include\\esp_hw_support\\include", "C:\\Users\\massi\\.platformio\\packages\\framework-arduinoespressif32\\tools\\sdk\\esp32\\include\\esp_hw_support\\include\\soc", "C:\\Users\\massi\\.platformio\\packages\\framework-arduinoespressif32\\tools\\sdk\\esp32\\include\\esp_hw_support\\include\\soc\\esp32", "C:\\Users\\massi\\.platformio\\packages\\framework-arduinoespressif32\\tools\\sdk\\esp32\\include\\esp [... cut ...] oup": "Platform"}], "extra": {"flash_images": [{"offset": "0x1000", "path": "C:\\Users\\massi\\AppData\\Local\\Temp\\pio-build\\ESP32_Gong-d06c40cc76\\wemos_d1_mini32\\bootloader.bin"}, {"offset": "0x8000", "path": "C:\\Users\\massi\\AppData\\Local\\Temp\\pio-build\\ESP32_Gong-d06c40cc76\\wemos_d1_mini32\\partitions.bin"}, {"offset": "0xe000", "path": "C:\\Users\\massi\\.platformio\\packages\\framework-arduinoespressif32\\tools\\partitions\\boot_app0.bin"}], "application_offset": "0x10000"}}
```

ivankravets, 2024-02-14:
If you comment `build_dir` in the `platformio.ini` and restart VSCode - does it work?

mbariola, 2024-02-14:
@ivankravets  Hi, yes when commenting that line (and workspace_dir too) the error no longer presents itself and the includes are found

ivankravets, 2024-02-15:
I can't reproduce this issue. Please ensure that the `TEMP` environment is set. You can check it with 
```
python -c "import os; print(os.environ)"
```

See the screenshot from my Windows VM:
[image]

closed 2024-02-15 by ivankravets

mbariola, 2024-02-15:
hi @ivankravets , you mean the TEMP env variable? yes it's set
among the big amount of other set stuff:
`'TEMP': 'C:\\Users\\massi\\AppData\\Local\\Temp', 'TMP': 'C:\\Users\\massi\\AppData\\Local\\Temp', `

Elektrik1, 2025-06-22:
Same error here, when I change workspace_dir to anything. In my case I set it as follows in platformio.ini:
`[platformio]
workspace_dir = ${sysenv.TEMP}/pio-workspaces/${PROJECT_HASH}`

bigwave, 2025-08-27:
I've updated my issue with more investigations:
https://github.com/platformio/platformio-core/issues/4754#issuecomment-3228176448